ethz-asl/voxblox #92: Fix segfaults with MergedTsdfIntegrator.

fabianbl, 2017-08-04:
Occasionally, I get segfaults after the latest voxblox update.. Any ideas?

```bash
Progress:   0.2% [--------------------------------------------------]I0804 16:13:05.733964 12470 tsdf_integrator.h:312] Went from 67 points to 41 raycasts  and 0 clear rays.
*** Error in `/home/fabian/Code/catkin_ws/devel/lib/voxblox_interface/test_voxblox_interface': double free or corruption (!prev): 0x000000000210e8e0 ***
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x777e5)[0x7fa767b007e5]
/lib/x86_64-linux-gnu/libc.so.6(+0x8037a)[0x7fa767b0937a]
/lib/x86_64-linux-gnu/libc.so.6(cfree+0x4c)[0x7fa767b0d53c]
/home/fabian/Code/catkin_ws/devel/lib/libvoxblox.so(_ZNSt6vectorIN7voxblox6timing13TimerMapValueESaIS2_EE19_M_emplace_back_auxIJS2_EEEvDpOT_+0xff)[0x7fa76744143f]
/home/fabian/Code/catkin_ws/devel/lib/libvoxblox.so(+0x38e62)[0x7fa767440e62]
/home/fabian/Code/catkin_ws/devel/lib/libvoxblox.so(_ZN7voxblox6timing5TimerC2ERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEb+0x1e)[0x7fa767440fbe]
/home/fabian/Code/catkin_ws/devel/lib/libvoxblox_interface.so(_ZN7voxblox20MergedTsdfIntegrator14integrateVoxelERKN5kindr7minimal26QuatTransformationTemplateIdEERKSt6vectorIN5Eigen6MatrixIdLi3ELi1ELi0ELi3ELi1EEENS8_17aligned_allocatorISA_EEERKS7_INS_5ColorESaISG_EEbbRKSt4pairINS9_IiLi3ELi1ELi0ELi3ELi1EEES7_ImSaImEEERKSt13unordered_mapISM_SO_NS_14BlockIndexHashESt8equal_toISM_ENSB_ISL_IKSM_SO_EEEEPSt5queueINS0_9VoxelInfoESt5dequeIS13_SaIS13_EEE+0x5ef)[0x7fa7699789af]
/home/fabian/Code/catkin_ws/devel/lib/libvoxblox_interface.so(_ZN7voxblox20MergedTsdfIntegrator15integrateVoxelsERKN5kindr7minimal26QuatTransformationTemplateIdEERKSt6vectorIN5Eigen6MatrixIdLi3ELi1ELi0ELi3ELi1EEENS8_17aligned_allocatorISA_EEERKS7_INS_5ColorESaISG_EEbbRKSt13unordered_mapINS9_IiLi3ELi1ELi0ELi3ELi1EEES7_ImSaImEENS_14BlockIndexHashESt8equal_toISM_ENSB_ISt4pairIKSM_SO_EEEESY_PSt5queueINS0_9VoxelInfoESt5dequeIS10_SaIS10_EEEm+0x1a8)[0x7fa769979448]
/home/fabian/Code/catkin_ws/devel/lib/libvoxblox_interface.so(_ZNSt6thread5_ImplISt12_Bind_simpleIFSt7_Mem_fnIMN7voxblox20MergedTsdfIntegratorEFvRKN5kindr7minimal26QuatTransformationTemplateIdEERKSt6vectorIN5Eigen6MatrixIdLi3ELi1ELi0ELi3ELi1EEENSC_17aligned_allocatorISE_EEERKSB_INS3_5ColorESaISK_EEbbRKSt13unordered_mapINSD_IiLi3ELi1ELi0ELi3ELi1EEESB_ImSaImEENS3_14BlockIndexHashESt8equal_toISQ_ENSF_ISt4pairIKSQ_SS_EEEES12_PSt5queueINS4_9VoxelInfoESt5dequeIS14_SaIS14_EEEmEEPS4_S8_SH_SM_bbS10_S10_S19_mEEE6_M_runEv+0x64)[0x7fa769973544]
/usr/lib/x86_64-linux-gnu/libstdc++.so.6(+0xb8c80)[0x7fa768121c80]
/lib/x86_64-linux-gnu/libpthread.so.0(+0x76ba)[0x7fa76974c6ba]
/lib/x86_64-linux-gnu/libc.so.6(clone+0x6d)[0x7fa767b903dd]
======= Memory map: ========
00400000-00417000 r-xp 00000000 103:02 1049832                           /home/fabian/Code/catkin_ws/devel/lib/voxblox_interface/test_voxblox_interface
00617000-00618000 r--p 00017000 103:02 1049832                           /home/fabian/Code/catkin_ws/devel/lib/voxblox_interface/test_voxblox_interface
00618000-00619000 rw-p 00018000 103:02 1049832                           /home/fabian/Code/catkin_ws/devel/lib/voxblox_interface/test_voxblox_interface
00c79000-02182000 rw-p 00000000 00:00 0                                  [heap]

```

fabianbl, 2017-08-04:
Actually happens only with the MergedTsdfIntegrator, switched to the SimpleTsdfIntegrator now.

Could still be an issue though?

ZacharyTaylor, 2017-08-04:
I just checked through my changes and the only one I can see that really affects the merged integrator, but I don't see how this would cause this. Do you have a bag that reliably segfaults you could give me as this would speed up tracking it down.

ZacharyTaylor, 2017-08-04:
Also does the fast integrator also segfault?

fabianbl, 2017-08-04:
Actually there seems to be a problem with this timer here:

https://github.com/ethz-asl/voxblox/blob/master/voxblox/include/voxblox/integrator/tsdf_integrator.h#L393-L395

```bash
*** SIGABRT (@0x3e800003917) received by PID 14615 (TID 0x7f3574a44700) from PID 14615; stack trace: ***
    @     0x7f3596d35390 (unknown)
    @     0x7f359509d428 gsignal
    @     0x7f359509f02a abort
    @     0x7f35950df7ea (unknown)
    @     0x7f35950e837a (unknown)
    @     0x7f35950ec53c cfree
    @     0x7f3594a2043f _ZNSt6vectorIN7voxblox6timing13TimerMapValueESaIS2_EE19_M_emplace_back_auxIJS2_EEEvDpOT_
    @     0x7f3594a1fe62 voxblox::timing::Timing::GetHandle()
    @     0x7f3594a1ffbe voxblox::timing::Timer::Timer()
    @     0x7f3596f57e00 voxblox::MergedTsdfIntegrator::integrateVoxel()
    @     0x7f3596f58568 voxblox::MergedTsdfIntegrator::integrateVoxels()
    @     0x7f3596f525e4 std::thread::_Impl<>::_M_run()
    @     0x7f3595700c80 (unknown)
    @     0x7f3596d2b6ba start_thread
    @     0x7f359516f3dd clone
```

fabianbl, 2017-08-04:
I don't have a bag currently, just encountered it in a unit test in maplab...

fabianbl, 2017-08-09:
Should be solved with ethz-asl/voxblox#95

closed 2017-08-09 by fabianbl