ethz-asl/voxblox #94: max_ray_length_m not used properly with all integrators

skohlbr, 2017-08-05:
The `max_ray_length_m` param is used properly with the simple integrator, but the merged integrator ignores it. When for instance using VLP-16 clouds as input this becomes apparent really fast as integration time gets very slow with clouds containing points tens of meters away.

ZacharyTaylor, 2017-08-07:
I did some tests the other day and the merge integrator was definitely using it, I will check and see if anythings changed

skohlbr, 2017-08-07:
Took a look and created PR #97 to fix the issue. The parameter was indeed used, but only if `allow_clear` was set to `true`. Otherwise, it was ignored, which lead to all rays getting integrated regardless if they exceed the specified max distance (making things super slow when using high range LIDARs).

commit 14ffd10d (2017-08-08): Check for max_ray_length_m independent of allow_clear (fixes #94)
commit b647f28d (2017-08-08): Merge pull request #97 from tu-darmstadt-ros-pkg/fix_max_ray_length_merged_integrator

Check for max_ray_length_m independent of allow_clear (fixes #94)
closed 2017-08-08 by helenol