ethz-asl/voxblox #99: Hash collisions are super easy to generate

ZacharyTaylor, 2017-08-07:
Our hashing function has some issues:

Turns out that for all the values I tested if an index is (x, y, z) then hash(x, y, z) == hash(x, -y, -z)
This means we will see a massive amount of hash collisions when exploring near the origin

magehrig, 2017-08-07:
It seems that you are computing the [combined hash value](https://github.com/ethz-asl/voxblox/blob/0070fed6721ca2ea905a3ef0f3323931490c2ec9/voxblox/include/voxblox/core/block_hash.h#L19-L20) with the XOR operator.  The XOR operator is commutative. In addition to what you have already mentioned, it is also true that has(x,y,z) == has(x,z,y). XOR is probably the wrong choice when dealing with coordinates.

ZacharyTaylor, 2017-08-14:
I looked more into this and it seems while the above issue with negatives is true, we are highly unlikely to get more than two voxels with the same hash. This means things are still O(1) and as long as we check the index rather than the hash when comparing locations its not a big problem. Closing.

closed 2017-08-14 by ZacharyTaylor