ethz-asl/voxblox #144: esdf_slice was not generated in RVIZ

BryceeeeeChen, 2018-01-03:
Hello, thanks for making this useful library open-source...I just ran the cow_and_lady_dataset.launch file and the mesh data was generated perfectly. But the esdf_slice which should be displayed on XY plane as described in the paper did not show up in RVIZ. Could someone help me solve this problem? Thanks.
[image]

ZacharyTaylor, 2018-01-03:
Hi, what does your launchfile look like, the esdf calculation and data visualization can both be fairly computationally expensive so are disabled by default.
Make sure `generate_esdf` and `publish_slices` are both set to true.

BryceeeeeChen, 2018-01-04:
@ZacharyTaylor Thank you for the reply. I have already set these two values to "true", otherwise the esdf_slice topic would not even show up in RVIZ. This is my launch file. 

<launch>

   <node pkg="tf" type="static_transform_publisher" name="link1_broadcaster" args="0 0 0 0 0 0 1 map world 100" />

  <arg name="play_bag" default="true" />
  <arg name="bag_file" default="/media/bxc237/HD-EG5U/data.bag"/>
  <arg name="voxel_size" default="0.05"/>
  <arg name="generate_esdf" default="true" />

  <node name="player" pkg="rosbag" type="play" output="screen" args="-r 1.0 -s 10 --clock $(arg bag_file)" if="$(arg play_bag)"/>

   <node name="voxblox_node" pkg="voxblox_ros" type="voxblox_node" output="screen" args="-alsologtostderr" clear_params="true">
    <remap from="pointcloud" to="/camera/depth_registered/points"/>
    <param name="tsdf_voxel_size" value="$(arg voxel_size)" />
    <param name="tsdf_voxels_per_side" value="16" />
    <param name="voxel_carving_enabled" value="true" />
    <param name="color_mode" value="color" />
    <param name="publish_slices" value="true" />
    <param name="use_tf_transforms" value="false" />
    <param name="update_mesh_every_n_sec" value="1.0" />
    <param name="min_time_between_msgs_sec" value="0.0" />
    <param name="method" value="fast" />
    <param name="generate_esdf" value="$(arg generate_esdf)" />
    <param name="use_const_weight" value="false" />
    <param name="allow_clear" value="false" />
    <param name="verbose" value="true" />
    <param name="max_integration_time_s" value="0.02"/>
    <param name="clear_checks_every_n_frames" value="1"/>
    <remap from="transform" to="/kinect/vrpn_client/estimated_transform" />
    <rosparam file="$(find voxblox_ros)/cfg/cow_and_lady.yaml"/>
    <param name="mesh_filename" value="$(find voxblox_ros)/mesh_results/$(anon cow).ply" />
  </node>

</launch>

BryceeeeeChen, 2018-01-04:
[cow_and_lady_dataset.launch.txt](https://github.com/ethz-asl/voxblox/files/1602137/cow_and_lady_dataset.launch.txt)

ZacharyTaylor, 2018-01-04:
Okay I was able to replicate the issue and it turns out its a bug that shows up when float(voxel_size) > double(voxel_size) and a layer slice is requested exactly between two blocks. This is the case for voxel_size = 0.05 and slice_level = 0.5 such as in the example.

I have pushed a quick fix for it, though unfortunately it creates a double layer in these instances. See https://github.com/ethz-asl/voxblox/pull/145

Also in your launchfile you might want to get rid of the line 
`<param name="max_integration_time_s" value="0.02"/>`
as this is quite a low limit and may cause quite a few voxels to be dropped, especially for smaller voxel sizes. I put this low limit in the main launchfile while testing and forgot to remove it before merging the changes. I will remove it in my next pr

BryceeeeeChen, 2018-01-04:
Thanks I can see the esdf_slice in RVIZ now. So I'm closing this issue about displaying ESDF.....I have not dug into the code yet. What is the possible cause for the double layer?

closed 2018-01-04 by BryceeeeeChen

ZacharyTaylor, 2018-01-04:
The slice code works by displaying all voxels who are within (voxel_size / 2) of the specified height. For these settings there are voxels at a height of 0.475 and 0.525, so in requesting a slice at 0.5 you end up exactly between the two voxels and end up getting both.

BryceeeeeChen, 2018-01-04:
Thank you very much for the explanation!