ethz-asl/voxblox #154: How to improve the clearing performance on the edge of view?

BryceeeeeChen, 2018-02-26:
Hi, the moving object would easily remain on the borderlines of point clouds when I generated the TSDF data ("allow_clear" is "true"). Is there a way to solve this problem? Thanks!

ZacharyTaylor, 2018-02-26:
Increasing the mesh_min_weight can prevent the system from outputting unstable areas such as are found on dynamic objects. The issue will be happening on the edge of the pointcloud as no new observations pass through these areas to clear the regions previously occupied by the dynamic object.

BryceeeeeChen, 2018-02-27:
Thank you. But with "allow_clear" set to "true", shouldn't the program clear every old obstacle no longer present within max_ray_length_m, even it can not raytrace something beyond those occupied voxels?

ZacharyTaylor, 2018-02-27:
All "allow_clear" does is allow the system to use points that are further away then "max_ray_length_m" for clearing. The clearing of rays shorter then this is governed by the "voxel_carving_enabled" flag which is true by default.

The system will not clear any object unless it receives information that the object is no longer there. That is it must observe a point in the environment which would not be visible if the dynamic object had not moved.

If you wish for a more general type of clearing behavior you could set the parameter "use_freespace_pointcloud" to true and then publish points that are behind the areas you want to clear to the "freespace_pointcloud" topic. This freespace topic works by sending clearing rays from the current camera origin to just before the truncation distance of the given freespace points.

BryceeeeeChen, 2018-03-01:
Thank you very much @ZacharyTaylor , I will do more tests based on your suggestion. Close the issue now.

closed 2018-03-01 by BryceeeeeChen