ethz-asl/voxblox #166: Documentation available?

Seanmatthews, 2018-04-12:
Where can I find an API reference or usage examples?

ZacharyTaylor, 2018-04-12:
The honest answer is that there currently really isn't any more documentation then you see here.
As far as usage examples go it depends a bit on your application:

If you just want to use voxblox as a ros node, you can look at one of the example launch files https://github.com/ethz-asl/voxblox/tree/master/voxblox_ros/launch
we have setups for the euroc dataset, the kitti dataset, our own and a few other use cases.

To see how things are hocked up under the hood, have a look at the esdf_sever, tsdf_server or simulation _server files in voxblox ros.

For an example of voxblox being used as a library you can check out the voxblox_interface that exists in maplab https://github.com/ethz-asl/maplab/tree/master/interfaces/voxblox_interface

ZacharyTaylor, 2018-11-19:
Its been a long time coming but voxblox is finally documented see:
https://voxblox.readthedocs.io/en/latest/api/library_root.html

closed 2018-11-19 by ZacharyTaylor

gogojjh, 2022-04-01:
Voxblox involves lots of operations on TSDF: integration, interpolation, gradient computation, and distance computation. Could you please recommend some papers or resources? T
Thanks!