ethz-asl/voxblox #170: Loop closure

Changliu52, 2018-04-17:
Hi,

Thanks for sharing the code.
I wonder if this does loop closure?

Kind Regards,
Chang

ZacharyTaylor, 2018-04-17:
Voxblox is a pure mapping framework and assumes that the poses are provided by another separate framework (We have used it with Rovio, Orb-slam, Vicon, etc). 

To ensure consistent maps we have performed some work to establish a method via which voxblox can be used online with frameworks that employ loop closures (https://arxiv.org/abs/1710.07242) however this code is currently not public.

Alternatively the maplab framework (https://github.com/ethz-asl/maplab) has the ability to not only provide state estimation but also do post-processing steps such as loop closure (albeit offline). Maplab has a voxblox interface to allow these poses to be used to reconstruct a dense map.

Changliu52, 2018-04-17:
Awesome. Thank you very much for the info.

closed 2018-04-17 by Changliu52

dario-ljubic, 2018-06-05:
Hello @ZacharyTaylor , 

First of all, thank You for sharing the code!
Question I have is closely related to this one so I am posting it here.
You said that code supporting loop closures is currently not public, but do You plan to release it?

Thank You very much!
King regards

alexmillane, 2018-06-06:
Hi @dario-ljubic 

Thanks for the interest. We do plan to make it public. However, this wont happen for at least a few months as the paper is currently under review and the code still needs a bit of work to make it usable by a general audience. It is coming though!

Cheers,
Alex

dario-ljubic, 2018-06-06:
Thank You for the quick answer!

I am looking forward to it!
Best regards

alexcherpi, 2019-05-17:
Hi @alexmillane ,
I'm interested in using Voxblox with a third party localization system (as it is not provided by Voxblox). But the loop closure is still missing. Have you already released the improved Voxblox embedding this loop closure? If no what do you recommend in the meantime (my system is provided with Depth-map and IMU)?
Best, Alex

alexmillane, 2019-05-17:
What is best to use depends on the specifics of your sensor setup. When you say depth-map, do you mean an RGB-D camera? Is this system hardware time-synchronized to the imu?

alexcherpi, 2019-05-20:
By depth map I mean a pointcloud. The IMU measurements can be time-synchronized.

alexmillane, 2019-05-20:
Is the pointcloud is coming from from a 3D lidar? If you're looking for a complete lidar mapping system I would recommend using something like Cartographer, one of the Loam variants, or Segmap.

alexcherpi, 2019-05-20:
The pointcloud is coming from a Stereo-Camera similar to the Realsense D435i

alexmillane, 2019-05-20:
Our approach is to use visual-inertial tracking on a sensor like that and then just project the depth information into a map voxblox/c-blox based map. In this way we ignore the 3D information from depth during camera tracking but, for us, this is fine. What exactly you use for localization depends on your requirements.

alexcherpi, 2019-06-16:
Hi @alexmillane ,
You said that Voxblox could be used with Cartographer. Have you done some tests with both running together on an onboard computer? If yes which was the onboard computer and what peformance did you obtain?

alexmillane, 2019-06-18:
I'm using that setup for my research, and so far only in an offline setting. On-board our drones we use Rovio (https://github.com/ethz-asl/rovio) and voxblox.

alexcherpi, 2019-06-18:
The issue is that my drone could fly in environment with a very small amount of visual features. Thus using ROVIO could be not robust enough. That's why I want to use RGB-D input instead of monocular images. Do you know an equivalent of ROVIO for RGB-D data (which runs onboard) ?

alexmillane, 2019-06-19:
You could have a look at DVO:

https://vision.in.tum.de/data/software/dvo

but honestly flying with and rgb-d camera in a visually degraded environment is a bit out of my expertise. What kind of environment are you talking about exactly?