ethz-asl/voxblox #173: Use ESDF for planning?

BryceeeeeChen, 2018-05-09:
Hi, are there any examples that I can follow to actually use voxblox for robot planning task (maybe use ESDF to replace costmap in ROS navigation stack or integrate it with existed planner interface)? Thanks!

helenol, 2018-06-05:
Yes, this is currently missing... I will shortly open-source some of my planners with examples. I've been getting this question a lot. ;)

The short answer in the mean-time is this, copied from an e-mail:
The way I use voxblox in ROS for planning is fairly simple: one esdf_server/voxblox_node runs and incorporates the new sensor data and builds an ESDF. Then this ESDF is published a full map message to a planner, which has a class member of type EsdfServer and listens to this topic.

Map type:
https://github.com/ethz-asl/voxblox/blob/master/voxblox_msgs/msg/Layer.msg
Complete map publisher/subscribers:
https://github.com/ethz-asl/voxblox/blob/master/voxblox_ros/src/esdf_server.cc#L29-L34
Param for publishing map:
https://github.com/ethz-asl/voxblox/blob/master/voxblox_ros/src/esdf_server.cc#L76

Then in the planners, we just make queries to the EsdfMap (contained by the EsdfServer) to get distances and gradients at specific points:
https://github.com/ethz-asl/voxblox/blob/master/voxblox/include/voxblox/core/esdf_map.h#L65-L77

BryceeeeeChen, 2018-06-09:
@helenol , thank you for the detailed explanation. The integration does look like a quite straightforward process and I will try it. Close issue.

closed 2018-06-09 by BryceeeeeChen