ethz-asl/voxblox #175: Does the project support Velodyne HDL-64E and odom as source?

bryanibit, 2018-06-11:
Velodyne HDL-64E produces a topic (type: sensor_msgs::PointCloud2) and odom can produce a topic (type: nav_msgs/Odometry). And the transformation between them is showed in tf topic. I wonder the Voxblox could support the data to produce surface mesh.

ZacharyTaylor, 2018-06-11:
We have used Velodyne HDL-64E data before in the KITTI dataset with fairly good results. There are a few parameters with defaults suited to rgbd cameras you will want to change however,

* `max_ray_length_m` - change to something like 20 or 30 meters.
* `min_ray_length_m` - if your vehicle is in the lidar fov change to something like 2 meters to exclude it.
* `use_const_weight` - lidar accuracy dosen't depend on distance so set this true.
* `truncation_distance` - a large truncation distance maybe 1 or 2 meters helps pick up the road as the lidar always hits it at a pretty shallow angle.
* `method` - setting to fast can help with the large number of voxels that need integrating.

Finally if performance is an issue try disabling voxel carving.

One final issue is that lidar maps can produce very large meshes. I usually found that rviz would begin lagging and having serious issues once the mesh exceeded a few million triangles.

bryanibit, 2018-06-11:
To be honest, I just want to merge all pointclouds from HDL-64E according to the position and azimuth of every frame of pointclouds through removing redundant pointcloud. So that the whole mosaic map looks like uniform for point distribution. I think it's an overkill to put your project on my job. Do you have any suggestion? Thank you so much!

ZacharyTaylor, 2018-06-11:
PCL has this style of thinning built into it. Either that or check out LibPointMatcher

helenol, 2018-06-18:
Are you referring to `Failed to find match for field 'rgb'.`? That's fine, your mesh just won't have color (which makes sense since you don't have color data).
Looks like it's successfully integrating data in if the calls show up in the timing. Be sure to modify the parameters @ZacharyTaylor said above for best results, and maybe try out with larger voxels to start.

bryanibit, 2018-06-18:
@helenol I updated my launch file to the following:

```
<launch>
<arg name="voxel_size" default="0.05"/>
   <node name="voxblox_node" pkg="voxblox_ros" type="tsdf_server" output="screen" args="-alsologtostderr" clear_params="true">
    <remap from="pointcloud" to="/velodyne_top_driver/velodyne_points"/>
    <param name="tsdf_voxel_size" value="$(arg voxel_size)" />
    <param name="tsdf_voxels_per_side" value="16" />
    <param name="voxel_carving_enabled" value="true" />
    <param name="color_mode" value="height" />
    <param name="use_tf_transforms" value="true" />
    <param name="update_mesh_every_n_sec" value="1.0" />
    <param name="min_time_between_msgs_sec" value="0.0" />
    <param name="method" value="fast" />
    <param name="use_const_weight" value="true" />
    <param name="allow_clear" value="true" />
    <param name="verbose" value="true" />
	<param name="max_ray_length_m" value="25.0" />
	<param name="min_ray_length_m" value="2.0" />
	<param name="truncation_distance" value="1.5" />
	<param name="generate_mesh" value="true" />
    <remap from="transform" to="/tf" />
    <rosparam file="$(find voxblox_ros)/cfg/cow_and_lady.yaml"/>
    <param name="mesh_filename" value="$(find voxblox_ros)/mesh_results/cow.ply" />
  </node>
</launch>
```

I changed the transform to the topic /tf, which included the transformation of frame_id of laser and vehicle position. And add some other params about voxblox, Howerer that still did not work, I think. Nothing still showed in rviz. I used *rostopic hz/echo /voxblox_node/tsdf_pointcloud*, then terminal showed **no new messages**.

ZacharyTaylor, 2018-06-18:
There seems to be a slight misunderstanding about how voxblox finds which transform to apply.

If "use_tf_transforms" is set to "false", voxblox will subscribe to a topic "transform" which should contain messages of type "geometry_msgs::TransformStamped"

If "use_tf_transforms" is set to "true", voxblox will subscribe to "/tf" and use a "tf::TransformListener" to find the transformation from a frame called "world" to the frame specified in the input pointcloud message's header. If you need either of these frames to have different names you can set them using the "world_frame" or "sensor_frame" parameters respectively. 

I am guessing the data you are inputting does not have a "world" tf frame and you just need to either change the "world_frame" parameter or create an appropriate static_transform_publisher node.

bryanibit, 2018-06-19:
@ZacharyTaylor I think I have set a frame called "world" before, using the following words.

```
<node name="$(arg LiDAR_top_id)_static_tf_node" pkg="static_tf_publisher" type="static_tf_publisher_clock_node" 	args="0.0 0.0 0.75 0.0 0.0 1.57 /base_link /$(arg LiDAR_top_id) 50"/>
<node name="map_static_tf_node" pkg="static_tf_publisher" type="static_tf_publisher_clock_node" args="0.0 0.0 0.0 0.0 0.0 0.0 1.0 /map /odom 50"/>
<node name="world_static_tf_node" pkg="static_tf_publisher" type="static_tf_publisher_clock_node" args="0.0 0.0 0.0 0.0 0.0 0.0 1.0 /world /map 50"/>
```

 The whole frame relationship is described to [frame.png](https://drive.google.com/open?id=1GK98o3eG-ggMZv9Q1Ve-W00CI0jWyF8f). And the node relationship is described [rosgraph.png](https://drive.google.com/open?id=1sH6o5UseP2nZoz-1hm3shMyrA9j7lqvn).

ZacharyTaylor, 2018-06-19:
This does all look correct to me. Are you getting any output from voxblox, is it saying its integrating frames or having any warnings / errors?

Also does "rostopic hz/echo /voxblox_node/mesh" also output nothing?

bryanibit, 2018-06-19:
@ZacharyTaylor /voxblox_node/mesh has messages, and I saw them in rviz, but tsdf_pointcloud and others have no messages. The computer becomes very very slow. To be honest, I do not need mesh. I just want occupancy grid for the geographical space and count the number or intensity of all points for each grid.

ZacharyTaylor, 2018-06-19:
Also if you are only needing a voxelized pointcloud, have you considered just using something like http://wiki.ros.org/pcl_ros/Tutorials/VoxelGrid%20filtering instead of a full TSDF pipeline?

bryanibit, 2018-06-19:
@ZacharyTaylor That makes sense. I have seen the tsdf_pointcloud and mesh like [those things](https://drive.google.com/file/d/1bsJnw-yAREKCBobVcBV0si-xPC96LEC4/view?usp=sharing
https://drive.google.com/file/d/1n6g962p2V3eGBqjRyDvP1QVkVC2NDfPU/view?usp=sharing
https://drive.google.com/file/d/1CnaaA5vmDzzzjUt16WXtylHADpTYQ5Gp/view?usp=sharing
https://drive.google.com/file/d/1x1H7AqSwRBGGC_-CcbShhAEDZZkunhtH/view?usp=sharing). I will try what you told me. The following link includes my launch file. Thank you so much for your patient!

closed 2018-08-22 by ZacharyTaylor