ethz-asl/voxblox #177: Catkin build error!

bryanibit, 2018-06-16:
I follow the tutorial and everything is normal but error happened when catkin build voxblox_ros. The following is the detail. Is there anyone who can help me? Thank you so much!

`  src catkin build voxblox_ros
-------------------------------------------------------
Profile:                     default
Extending:        [explicit] /opt/ros/kinetic
Workspace:                   /home/bryan/ros_ws
-------------------------------------------------------
Source Space:       [exists] /home/bryan/ros_ws/src
Log Space:         [missing] /home/bryan/ros_ws/logs
Build Space:        [exists] /home/bryan/ros_ws/build
Devel Space:        [exists] /home/bryan/ros_ws/devel
Install Space:      [unused] /home/bryan/ros_ws/install
DESTDIR:            [unused] None
-------------------------------------------------------
Devel Space Layout:          merged
Install Space Layout:        None
-------------------------------------------------------
Additional CMake Args:       -DCMAKE_BUILD_TYPE=Release
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
-------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
-------------------------------------------------------
Workspace configuration appears valid.

NOTE: Forcing CMake to run for each package.
-------------------------------------------------------
[build] Found '12' packages in 0.0 seconds.                                                                                                                                                                
[build] Updating package table.                                                                                                                                                                            
Starting  >>> catkin_simple                                                                                                                                                                                
Finished  <<< catkin_simple                       [ 1.5 seconds ]                                                                                                                                          
Starting  >>> eigen_catkin                                                                                                                                                                                 
Starting  >>> gflags_catkin                                                                                                                                                                                
Starting  >>> voxblox_msgs                                                                                                                                                                                 
Finished  <<< eigen_catkin                        [ 1.8 seconds ]                                                                                                                                          
Finished  <<< voxblox_msgs                        [ 3.0 seconds ]                                                                                                                                          
___________________________________________________________________________________________________________________________________________________________________________________________________________
Warnings   << gflags_catkin:make /home/bryan/ros_ws/logs/gflags_catkin/build.make.000.log                                                                                                                  
make[3]: warning: -jN forced in submake: disabling jobserver mode.
make[3]: warning: -jN forced in submake: disabling jobserver mode.
cd /home/bryan/ros_ws/build/gflags_catkin; catkin build --get-env gflags_catkin | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
...........................................................................................................................................................................................................
Finished  <<< gflags_catkin                       [ 13.4 seconds ]                                                                                                                                         
Starting  >>> glog_catkin                                                                                                                                                                                  
___________________________________________________________________________________________________________________________________________________________________________________________________________
Errors     << glog_catkin:make /home/bryan/ros_ws/logs/glog_catkin/build.make.000.log                                                                                                                      
/bin/sh: 1: autoreconf: not found
make[2]: *** [glog_src-prefix/src/glog_src-stamp/glog_src-configure] Error 127
make[1]: *** [CMakeFiles/glog_src.dir/all] Error 2
make: *** [all] Error 2
cd /home/bryan/ros_ws/build/glog_catkin; catkin build --get-env glog_catkin | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
...........................................................................................................................................................................................................
Failed     << glog_catkin:make                    [ Exited with code 2 ]                                                                                                                                   
Failed    <<< glog_catkin                         [ 5.9 seconds ]                                                                                                                                          
Abandoned <<< eigen_checks                        [ Unrelated job failed ]                                                                                                                                 
Abandoned <<< minkindr                            [ Unrelated job failed ]                                                                                                                                 
Abandoned <<< minkindr_conversions                [ Unrelated job failed ]                                                                                                                                 
Abandoned <<< voxblox                             [ Unrelated job failed ]                                                                                                                                 
Abandoned <<< voxblox_rviz_plugin                 [ Unrelated job failed ]                                                                                                                                 
Abandoned <<< voxblox_ros                         [ Unrelated job failed ]                                                                                                                                 
[build] Summary: 4 of 11 packages succeeded.                                                                                                                                                               
[build]   Ignored:   1 packages were skipped or are blacklisted.                                                                                                                                           
[build]   Warnings:  1 packages succeeded with warnings.                                                                                                                                                   
[build]   Abandoned: 6 packages were abandoned.                                                                                                                                                            
[build]   Failed:    1 packages failed.                                                                                                                                                                    
[build] Runtime: 21.0 seconds total.                                                                                                                                                                       
[build] Note: Workspace packages have changed, please re-source setup files to use them.
`

ZacharyTaylor, 2018-06-16:
The important line in your build file is   
`/bin/sh: 1: autoreconf: not found`  

It looks like we missed a dep in our install instructions, I have updated them now. You can fix this by going  
`sudo apt-get install autoconf`

bryanibit, 2018-06-17:
Thank you for your help! That really helped me. I am sorry that I ignored the all dependency of the project.

closed 2018-06-17 by bryanibit