ethz-asl/voxblox #180: Stereo Dense Reconstruction Nodes

shravank1996, 2018-06-24:
I am trying to run EuRoC dataset for VoxBlox. I am getting an error stating stereo dense reconstruction not found? From where is the package stereo dense reconstruction nodes being used? Is it possible to get a link to the same? or is it the one from https://github.com/umass-amrl/stereo_dense_reconstruction

ZacharyTaylor, 2018-06-24:
Its the official ros one http://wiki.ros.org/stereo_image_proc

shravank1996, 2018-06-24:
The launch file for EuRoC_dataset_bag has a node called stereo_dense_reconstruction from a package called stereo_dense_reconstruction_nodes, but there isnt a package named that in stereo_image_proc.

Am I missing something here?

Thanks

ZacharyTaylor, 2018-06-24:
My apologies, I read your comment too fast and thought you were talking about the KITTI launchfile, which uses stereo_image_proc. 

Unfortunately stereo_dense_reconstruction is not open source at the moment. I will look at changing the launchfile to use stereo_image_proc

helenol, 2018-09-19:
This is resolved now, all remaining launch files use either standard ROS packages or image_undistort.

closed 2018-09-19 by helenol