ethz-asl/voxblox #181: Using Voxblox in planners that do not support ESDF

dario-ljubic, 2018-07-03:
Hello, 
first of all, thank you on making this excellent code public!

I am currently experimenting with the Voxblox and looking into its possibilities. The recent change on the possibility of local scene reconstruction looks and works great.
Now, I would also like to use it in my planner, but it does not operate with ESDF. I have tried taking surface point cloud from TSDF and using point cloud to laser scan converter which is able to create a 2D-like scan that I can use in my planner.

However, with the frequency of ~10 Hz, it is too slow. I was wondering do You maybe have an advice on what is the best way to convert TSDF/ESDF to laser scan or to obtain something like laser scan?

Thank You in advance!

ZacharyTaylor, 2018-07-03:
Thanks for the interest in the code. 
If "publish_tsdf_info" is set to true, then voxblox should output the surface pointcloud whenever the TSDF changes. 

So if you are only seeing output at 10 Hz there are four possibilities
* We have a bug
* The "min_time_between_msgs_sec" parameter has been set to ~0.1. In this case just set it to 0.0 to prevent it throttling your data.
* Your input data is coming in at 10 Hz. In this case publishing faster would not add any information
* Voxblox is overloaded and dropping messages. If this is happing try decreasing the max ray length or increasing voxel size.

For getting a laser like output your best options are either the surface pointcloud or the mesh_pcl topic. 
Why are you not just directly using the lidar output, is voxblox needed to denoise or downsample the data?

dario-ljubic, 2018-07-24:
Thank You for the reply!

Whenever I enable "publish_tsdf_info" it slows down everything quite a bit. Following your advices I have found out that the "voxel_size" was the problem.
Now, combining "voxel_size" and "max_block_distance_from_body" I get the frequency above 15 Hz.

Yes, that was the idea - to denoise and downsample data and to have a good reconstruction quality on the way. 
Also, I have seen the "mesh_pcl" topic you mentioned, but when I try to find "output_mesh_as_pcl_mesh" I only find it in readme files.