ethz-asl/voxblox #183: rviz can't visualize /voxblox_node/mesh

martinakos, 2018-07-12:
I found [this](https://robotics.stackexchange.com/questions/15547/rviz-how-to-visualize-a-voxblox-mesh-as-a-markerarray) unanswered question online.  I'm in the same situation, I can't visualize the mesh published by voxblox with rviz.

martinakos, 2018-07-12:
I wasn't sourcing the corresponding setup.bash in the terminal that runs rviz

closed 2018-07-12 by martinakos

ZacharyTaylor, 2018-07-12:
A little info as to why you need to source your workspace as the doc's were out of date.

The mesh is actually of type `voxblox_msgs::MeshBlock`. Voxblox use to output a standard MarkerArray, however this form of visualization means you have to output the entire mesh every time any part of it changes. This meant when running on a robot and visualizing on a laptop we were saturating our wifi connection to display even relatively small areas.

To correct this issue we made a custom message that supports partial updates only changing the mesh where needed. To use this custom message you have to build voxblox_rviz_plugin and have it in your workspace when running rviz.