ethz-asl/voxblox #187: How to load a pointcloud map that already existed?

Aryayay, 2018-08-02:
Hi ,

I want to continue mapping with prebuilt point cloud map, is this supported in your code?

ZacharyTaylor, 2018-08-02:
The loading and saving of voxel maps via protobuf is implemented and fully supported.
However, if you have a pointcloud that you wish to turn into a map this is slightly more difficult. The formation of TSDFs from pointclouds usually takes two inputs, the points and the location they were viewed from. In a pre-built pointcloud map this viewpoint information is missing and so some form of approximation to it would have to be generated (for example assuming points were viewed along their normals). This is not implemented but would not be too hard to add if there was sufficient interest in such a feature.

Aryayay, 2018-08-02:
ok, its a goog idea!  Thx a lot.

ZacharyTaylor, 2018-09-05:
Protobuf is a generic way of serializing data structures https://developers.google.com/protocol-buffers/
We use it for serialization of the TSDF map

ZacharyTaylor, 2018-09-05:
To go from a pointcloud to a TSDF you need two pieces of information.
1) which scan each point in the pointcloud originally belonged to
2) the pose in the world frame the sensor had when it made each scan

You can approximate this, but a simple combined pointcloud would not have enough information by itself to create a TSDF

plumewind, 2018-09-06:
Thank you very much for your reply, thank you, I understand now.

closed 2018-11-07 by ZacharyTaylor

joerowelll, 2022-11-21:
Has this been implemented in Voxblox?