ethz-asl/voxblox #189: How to view esdf in rviz

diyuhuodong, 2018-08-13:
Hi,I run roslaunch voxblox_ros cow_and_lady_dataset.launch.  no ESTF generated 
[image]
       I saw ESTF on YouTube. 
     
[image]
      Maybe I have some problems with configuration file。
    
<launch>
  <node pkg="tf" type="static_transform_publisher" name="link1_broadcaster" args="0 0 0 0 0 0 1 map world 100" />

  <arg name="play_bag" default="true" />
  <arg name="bag_file" default="/home/byd/vio/bag/data.bag"/>
  <arg name="voxel_size" default="0.1"/>
  <arg name="generate_esdf" default="true" />

  <node name="player" pkg="rosbag" type="play" output="screen" args="-r 1.0 -s 10 --clock $(arg bag_file)" if="$(arg play_bag)"/>

  <node name="voxblox_node" pkg="voxblox_ros" type="esdf_server" output="screen" args="-alsologtostderr" clear_params="true">
    <remap from="pointcloud" to="/camera/depth_registered/points"/>
    <param name="tsdf_voxel_size" value="$(arg voxel_size)" />
    <param name="tsdf_voxels_per_side" value="16" />
    <param name="voxel_carving_enabled" value="true" />
    <param name="color_mode" value="color" />
    <param name="publish_slices" value="true" />
    <param name="use_tf_transforms" value="false" />
    <param name="update_mesh_every_n_sec" value="1.0" />
    <param name="min_time_between_msgs_sec" value="0.0" />
    <param name="method" value="fast" />
    <param name="generate_esdf" value="$(arg generate_esdf)" />
    <param name="allow_clear" value="false" />
    <param name="use_const_weight" value="false" />
    <param name="publish_esdf_map" value="true" />
    <param name="publish_tsdf_map" value="true" />
    <param name="verbose" value="true" />
    <param name="max_integration_time_s" value="0.1"/>
    <param name="clear_checks_every_n_frames" value="1"/>
    <remap from="transform" to="/kinect/vrpn_client/estimated_transform" />
    <rosparam file="$(find voxblox_ros)/cfg/cow_and_lady.yaml"/>
    <param name="mesh_filename" value="$(find voxblox_ros)/mesh_results/$(anon cow).ply" />
 </node>
</launch>

diyuhuodong, 2018-08-13:
cow_and_lady_dataset.launch
<launch>
  <node pkg="tf" type="static_transform_publisher" name="link1_broadcaster" args="0 0 0 0 0 0 1 map world 100" />

  <arg name="play_bag" default="true" />
  <arg name="bag_file" default="/home/byd/vio/bag/data.bag"/>
  <arg name="voxel_size" default="0.1"/>
  <arg name="generate_esdf" default="true" />

  <node name="player" pkg="rosbag" type="play" output="screen" args="-r 1.0 -s 10 --clock $(arg bag_file)" if="$(arg play_bag)"/>

  <node name="voxblox_node" pkg="voxblox_ros" type="esdf_server" output="screen" args="-alsologtostderr" clear_params="true">
    <remap from="pointcloud" to="/camera/depth_registered/points"/>
    <param name="tsdf_voxel_size" value="$(arg voxel_size)" />
    <param name="tsdf_voxels_per_side" value="16" />
    <param name="voxel_carving_enabled" value="true" />
    <param name="color_mode" value="color" />
    <param name="publish_slices" value="true" />
    <param name="use_tf_transforms" value="false" />
    <param name="update_mesh_every_n_sec" value="1.0" />
    <param name="min_time_between_msgs_sec" value="0.0" />
    <param name="method" value="fast" />
    <param name="generate_esdf" value="$(arg generate_esdf)" />
    <param name="allow_clear" value="false" />
    <param name="use_const_weight" value="false" />
    <param name="publish_esdf_map" value="true" />
    <param name="publish_tsdf_map" value="true" />
    <param name="verbose" value="true" />
    <param name="max_integration_time_s" value="0.1"/>
    <param name="clear_checks_every_n_frames" value="1"/>
    <remap from="transform" to="/kinect/vrpn_client/estimated_transform" />
    <rosparam file="$(find voxblox_ros)/cfg/cow_and_lady.yaml"/>
    <param name="mesh_filename" value="$(find voxblox_ros)/mesh_results/$(anon cow).ply" />
 </node>
</launch>

ZacharyTaylor, 2018-08-13:
The key issue is you need to run a different node to get esdf information (note the same node still gives tsdf info)

In the launchfile change type="tsdf_server" to type="esdf_server"

DRAEYE, 2019-06-02:
Hello,I run with the same launch.txt, but cannot see any esdf output in rviz, how do you deal with that?

eyildiz-ugoe, 2019-06-12:
How can we visualize the colored mesh, like the ones shown in the main page? I think the authors need to make visualization more understandable. It's quite unclear.