ethz-asl/voxblox #191: No esdf point cloud with cow_and_lady_dataset

schnph, 2018-08-14:
Good day,

I did try the example data file cow_and_lady_dataset (data.bag)

I can see the slice of the esdf and esdf_map_out has messages on the ROS Topic.
On ROS Topic esdf_pointcloud is never a message.

How can I fix this?

here is my launchfile:
[cow_and_lady_dataset.txt](https://github.com/ethz-asl/voxblox/files/2286945/cow_and_lady_dataset.txt)

helenol, 2018-08-14:
Try adding the following parameters:

    <param name="publish_pointclouds" value="true" />
    <param name="publish_slices" value="true" />

Maybe we should consider changing the defaults... ;)

ZacharyTaylor, 2018-08-14:
You can either call the publish_pointclouds service, or set the parameter `publish_pointclouds` to `true`. This was fairly poorly documented and I have now updated the readme

ZacharyTaylor, 2018-08-14:
No problem, the publish_pointcloud flag was missing from the readme