ethz-asl/voxblox #196: Problem in visulizing topics in Rviz after 'source'

JillWangJill, 2018-08-18:
_Hi,  thanks a lot for providing such a useful tool.

I already `source devel/setup.bash.`  And changed the bag path in cow_and_lady_dataset.launch file.

I met problem in using rviz together with this package. My rviz fail to display the /voxblox_node/mesh
Here is something I am confused with and need your kind help.

1. My rviz looks like this.
[image]

2. my rviz terminal will show 

> [ WARN] [1534564039.729754290]: MessageFilter [target=camera_depth_frame ]: Dropped 100.00% of messages so far. Please turn the [ros.voxblox_rviz_plugin.message_notifier] rosconsole logger to DEBUG for more information.

3. If I echo /voxblox_node/mesh, 

> `      - 
        x: [0.875, 0.875, 0.8785830140113831]
        y: [-2.469820499420166, -2.4749999046325684, -2.4749999046325684]
        z: [2.375, 2.3868136405944824, 2.375]
        r: [139, 139, 139]
        g: [133, 133, 133]
        b: [130, 130, 130]
        a: [255, 255, 255]
--
header: 
  seq: 57
  stamp: 
    secs: 1534564190
    nsecs: 150633121
  frame_id: "world"
mesh_blocks: []
---

The mesh block is empty.

4.  in my roslaunch terminal,

> [ WARN] [1534564560.045967660]: No match found for transform timestamp: 1473521413.111484105
[ WARN] [1534564560.046022270]: Queue front: 1473521411.774098268 back: 1473521412.044124231

I am guessing the problem is in my rviz setting. In my rviz, the fixed frame is /camera/depth/frame. But I am not sure whether this is the problem and fail to know how to modify it.

Thanks a lot in advance and look forward!

JillWangJill, 2018-08-18:
I change Fixed frame into world by typing it. And I can see the mesh from rviz now. 

However, I am still confused about few questions.  

1. Why would there be WARN in roslaunch terminal. 

> No match found for transform timestamp. 
and 

>  Couldn't look up pose for incoming pointcloud.
happen time to time.

2. Even if I can see see mesh from rviz, the global status said Warn.

>  No tf data.  Actual error: Fixed Frame [world] does not exist
[image]

3. echo / voxblox_node/mesh still print same message in last comment, in which the mesh block is empty.

Sorry to ask these stupid questions but I really  need your generous help,   :  (   

Thanks a lot!

ZacharyTaylor, 2018-08-18:
1) Mesh blocks are sent out every time a section of the map changes. In many cases these blocks will only contain empty space, however we still send these blocks to rviz in case there was previously something meshed in this location that needs to be cleared from the display. As long as things are visualizing correctly this shouldn't be an issue.

Also Voxblox has a parameter called `world_frame` you can use to set the frame the map is published in.

2) When Voxblox goes to integrate a new pointcloud it will look up the transform from the corresponding time. If all of the transforms are older than the pointcloud, it will print a warning. Again, as long as you only get this now and then and not every frame it isn't a major issue.

JillWangJill, 2018-08-19:
Thanks a lot, @ZacharyTaylor .

Can I ask one more thing? I am trying to connect orbslam2_ros with voxblex together using one Kinect v1 as input.

As for the parameter file, I use file with different format from voxblox's. For instance, in orbslam2, the parameter file look like
[image]

But I am confused of how to merge these information into our format as follow,

> 
>  actually T_R_C (C = cam0, R = rgbd cam)
> T_B_C:
> - [1.0, 0.0, 0.0, 0.0]
> - [0.0, 1.0, 0.0, 0.0]
> - [0.0, 0.0, 1.0, 0.0]
> - [0.0, 0.0, 0.0, 1.0]
> invert_T_B_C: false
> 
>  actually T_V_C (C = cam0, V = vicon)
> T_B_D:
> - [0.971048, -0.120915, 0.206023, 0.00114049]
> - [0.15701, 0.973037, -0.168959, 0.0450936]
> - [-0.180038, 0.196415, 0.96385, 0.0430765]
> - [0.0, 0.0, 0.0, 1.0]
> invert_T_B_D: true

Although I know that intrinsic matrix's structure is like 

> fx s cx 0
> 0 fy cy 0
> 0 0 1 0
> 0 0 0 1

I fail to make it as one input of our format. Look forward to your reply and thx in advance!

ZacharyTaylor, 2018-08-19:
No problem. So if you are using the same sensor as both your source of depth and motion estimation than you don't actually need to set T_B_C or T_B_D (because they are both identity). These are only used if you have your tracking done in a different fixed frame to your depth information (for example tracking using a motion capture system as was done for the lady_and_cow dataset). 

Also for future reference these are 4x4 extrinsic transformations between the sensors and unrelated to the camera intrinsics.

JillWangJill, 2018-08-19:
Sincere thank again! @ZacharyTaylor 

I set TBC and TBD both into identity matrix. But when I try to save generated mesh as ply file, I failed.

Here is my launch file.

[try (copy).launch.txt](https://github.com/ethz-asl/voxblox/files/2300349/try.copy.launch.txt)

When I play`  rosservice info /voxblox_node/generate_mesh`
Terminal shows 
```
Node: /voxblox_node
URI: rosrpc://love-mua:52587
Type: std_srvs/Empty
Args: 
```

Could you give any hints? Really sorry to bother you so many times.

JillWangJill, 2018-08-19:
Sincere thank again! @ZacharyTaylor I set TBC and TBD both into identity matrix. 

I am trying to save generated mesh as ply file. I tried by adding  
> arg name="generate_mesh" default="true" /
into my launch file. 

But this service uses empty request and response, which means if I check, terminal show as follows.
When I play` rosservice info /voxblox_node/generate_mesh`
Terminal shows
```
Node: /voxblox_node
URI: rosrpc://love-mua:52587
Type: std_srvs/Empty
Args: 
```
`rossrv show std_srvs/Empty `
```
---

```
Could you give any hints on how to save the generated mesh instead of updating mesh? Really sorry to bother you so many times.

Here is my launch file.
[try (copy).launch.txt](https://github.com/ethz-asl/voxblox/files/2300355/try.copy.launch.txt)

JillWangJill, 2018-08-20:
Hi, @ZacharyTaylor  Thanks a lot!

I still fail to use generate_mesh service to save the mesh file.

Instead, I write one cpp to subscribe /voxblox_node/mesh topic to save it. But I fail to catkin build. The error message says
 ```
 error: __s_getMD5Sum is not a member of voxblox::Mesh
       return M::__s_getMD5Sum().c_str();
```
I think it is because ros fail to recognize this message type. But I donno how to fix it.

Below is my cmakelist file in voxblox_ros folder. And the cpp file I wrote.
[CMakeLists.txt](https://github.com/ethz-asl/voxblox/files/2302413/CMakeLists.txt)

Can you give any suggestions on this bug?
[meshsubscriber .cc.txt](https://github.com/ethz-asl/voxblox/files/2302416/meshsubscriber.cc.txt)

ZacharyTaylor, 2018-08-20:
@JillWangJill when you call the generate_mesh command the mesh will be saved as a ply file in the path set by `mesh_filename`. In your case this is set to `<param name="mesh_filename" value="$(find voxblox_ros)/mesh_results/$(anon cow).ply" />`.

However, I believe that the folder mesh_results doesn't exist by default and the output won't create a folder which is why it is failing. Either change the path of create the folder and it should work.

Also there is no need to write your own voxblox_mesh to ply mesh function. We already have them written and kept here https://github.com/ethz-asl/voxblox/tree/master/voxblox/include/voxblox/io

JillWangJill, 2018-08-21:
Dear @ZacharyTaylor ,

Thanks a lot! However, after I change the `mesh_filename` into absolute path, the saving procedure still fail.  After I use` ctrl+c` to close the terminal running `roslaunch voxblox_ros try.launch`, I cant find corresponding file in my folder. 
[try (copy).launch.txt](https://github.com/ethz-asl/voxblox/files/2304371/try.copy.launch.txt)

Help, please! Thanks in advance!

ZacharyTaylor, 2018-08-21:
Does voxblox output an error to the console, if so what does it say?

JillWangJill, 2018-08-21:
Here is the output in voxblox console.
[ouput.txt](https://github.com/ethz-asl/voxblox/files/2305250/ouput.txt)

helenol, 2018-08-21:
Did you call the generate_mesh service?
It should output either "Output file as PLY:" or "Failed to output mesh as PLY:" when you call the service in the console.

JillWangJill, 2018-08-21:
@helenol @ZacharyTaylor 
Thanks a lot!!!!!!!!!!!!!!!!!!!! 
I open another terminal and input `rosservice call /voxblox_node/generate_mesh "{}" `. The mesh is saved successfully.

Really sorry to bother so many times and thanks again!

helenol, 2018-08-22:
Glad it worked! :)

closed 2018-08-22 by helenol

joerowelll, 2022-11-27:
**Generate mesh not working and no specific error message.** 

I am having a similar problem in that when I call 

`rosservice call /voxblox_ros generate mesh "{}"`

The terminal outputs:

```[ INFO] [1669558636.376667689]: Updating mesh.
[ INFO] [1669558636.596380639]: Failed to output mesh as PLY: $PATH/mickie_ori_28672_303483_1396867906941177706.ply
[ INFO] [1669558636.596517636]: Mesh Timings: 
SM Timing
-----------
mesh/generate	      2	00.005219	(00.002609 +- 00.000137)	[00.002472,00.002746]
mesh/output  	      2	00.000028	(00.000014 +- 00.000010)	[00.000004,00.000025]
mesh/publish 	     72	00.001090	(00.000015 +- 00.000007)	[00.000007,00.000046]
mesh/update  	     70	00.123028	(00.001758 +- 00.000844)	[00.000852,00.004013]

[ INFO] [1669558637.376771854]: Updating mesh.
```

My launch file is as follows:
```
<launch>
  <arg name="play_bag" default="true" />
  <!-- <arg name="bag_file" default="/$PATH.bag"/> -->
  <arg name="voxel_size" default="0.05"/>

  <arg name="bag_file" default="$PATH.bag"/>
  <node name="player" pkg="rosbag" type="play" output="screen" args="-r 1.0 --clock $(arg bag_file)" if="$(arg play_bag)"/>

   <node name="voxblox_node" pkg="voxblox_ros" type="tsdf_server" output="screen" args="-alsologtostderr" clear_params="true">
    <remap from="pointcloud" to="/camera/depth_registered/points"/>
    <param name="tsdf_voxel_size" value="$(arg voxel_size)" />
    <param name="tsdf_voxels_per_side" value="16" />
    <param name="voxel_carving_enabled" value="true" />
    <param name="color_mode" value="color" />
    <param name="use_tf_transforms" value="false" />
    <param name="update_mesh_every_n_sec" value="1.0" />
    <param name="min_time_between_msgs_sec" value="0.0" />
    <param name="method" value="fast" />
    <param name="use_const_weight" value="false" />
    <param name="allow_clear" value="true" />
    <param name="verbose" value="true" />    
    <remap from="transform" to="/kinect/vrpn_client/estimated_transform" />
    <rosparam file="$(find voxblox_ros)/cfg/mickie.yaml"/>
    <param name="generate_mesh" value="true" />
    <param name="output_mesh_as_pcl_mesh" value="true" />
    <param name="mesh_filename" value="$(find voxblox_ros)/mesh_results/$(anon mickie6).ply" />
  </node>

</launch> 
```

The folder in which the mesh is saved exists, and it has worked with the cow and lady rosbag. Any ideas?