ethz-asl/voxblox #202: Decide what to do with pybindings and tango interface

helenol, 2018-08-22:
Since there's no longer a google tango, perhaps it doesn't make sense to keep them?
The python bindings can also move to their own package/repo, as we do not maintain them or test them, and they're anyway on the higher level interface (i.e., all they access is the internal ESDF layer anyway, no need for them to be in voxblox_ros).

@ZacharyTaylor @mfehr : thoughts?

ZacharyTaylor, 2018-08-22:
I had a browse through the forks of voxblox a while back and it seems like some people do use them. However, I agree that if we are not using / testing them we will eventually break them all. Not sure what the solution is

mfehr, 2018-08-27:
I think moving them to a separate package/repo makes sense, because without unit testing we will break them. Maybe it's worth trying to get in touch with some of the people using them, they might be willing to help/take over maintenance or even write some tests?

helenol, 2018-09-19:
Split out into https://github.com/ethz-asl/voxblox_deprecated/

closed 2018-09-19 by helenol