ethz-asl/voxblox #216: How to use save_map service ?

plumewind, 2018-09-19:
Hello, friends
      When I use the voxblox node, I want to save the TSDF. But when I enter the following instructions:

$  rosservice call /voxblox_node/save_map "test1.vxblx"

But nothing happened,  and I couldn't find the saved file. Is it stored in a strange location? Why can't I search for it?
And

$  rosservice type /voxblox_node/save_map | rossrv show
string file_path
---

Where am I wrong? Can someone help me? thank you very much
[image]

plumewind, 2018-09-19:
Sorry, I asked myself, I am stupid, I should understand it after think about it.

`rosservice call /voxblox_node/load_map ~/test1.vxblx`

closed 2018-09-19 by plumewind

rookierobot, 2025-03-12:
Hi @plumewind , can i ask how to use save_map service?  when i call this service, i meet: `ERROR: service [/voxblox_node/save_map] responded with an error: b''`, thank you for your reply