ethz-asl/voxblox #219: rivz plugin getting corrupted

griz1112, 2018-09-19:
Works fine for a few runs with the test datasets then the pointcloud won't show up in rviz. The topic isn't there. Think its the rviz plugin that is getting corrupted during shut down. If I do a catkin clean then rebuild it still doesn't work but If I create a new workspace and rebuild it does work.

helenol, 2018-09-19:
Never ever seen this issue before.
Sounds more like a problem with your rviz plugin loading (and rviz settings) rather than anything else. Have you tried cleaning your rviz settings? Paths to the plugin are stored there.

griz1112, 2018-09-19:
Its on a jetson xavier with the preview OS so I suspect its that if you haven't seen it before. I had another issue building something else where it couldn't write a file to .cmake even though the permissions owner group were right.  I've reinstalled ros-melodic and all that a couple of times. The last time I rebuilt and reinstalled i saved the default.rviz file before ever running rviz. I need to do more research on how rviz plugins work to get any further I think. Thanks

griz1112, 2018-09-19:
It was brain fade on my part :) Had two windows up one to start rviz and one to start voxblox. Silly me having not had any experience with rviz plugins figured I didn't need to source the setup.bash file in both windows. Sorry about that.

closed 2018-11-07 by ZacharyTaylor