ethz-asl/voxblox #226: True euclidean distance

Magnusgaertner, 2018-11-07:
I compared the distance errors in simulation using a cummulative histogram on the errors generated in simulation: 

[image]
https://plot.ly/~magnusgaertner/5/voxblox-euclidean-vs-voxblox/

Using the params: 
```
add_robot_pose: false
esdf_euclidean_distance: true/false (see graph)
esdf_min_distance_m: 0.2
generate_mesh: true
incremental: false
max_attempts_to_generate_viewpoint: 50
num_viewpoints: 50
truncation_distance: 0.4
tsdf_voxel_size: 0.1
tsdf_voxels_per_side: 16
verbose: true
world_frame: map
```

Only points that were behind the true surface were ignored for the evaluation. 
```
 ignore_behind_test_surface = false;
 ignore_behind_gt_surface = true;
```
Where are the bottom 30% in the esdf comming from? Is this a bug or do i miss something?

helenol, 2018-11-07:
@Magnusgaertner thanks for the evaluation!

Are you talking about the esdf_euclidean having more underestimation of the distance than non-euclidean distance?
I'm not 100% sure here... Could be a bug in the full euclidean distance, especially behind surface (to be honest I usually only evaluate outside of surface case, as this is what I care much more about for planning), or could be something up in the way the evaluations are done.
How do the results change if you ignore behind-surface, even for the test?

Magnusgaertner, 2018-11-08:
[image]

https://plot.ly/~magnusgaertner/5/issueeuclideanoutside-issueeuclideantrueoutside-issuenoeuclideanoutside-issuenoe/#/

Jep, the euclidean distance is seems to be systematically underestimated by a large amount (up to half a meter) and that not just for a few outliers but for almost 30% of the points this seems to be the case in some way. 

If you follow the link you can toggle the visibility of the different errors by clicking on the data lables  ;)
But basically the data does not change at all if we only allow both estimated and true distance to be positive.