ethz-asl/voxblox #227: CMake Error at  voxbloxConfig.cmake

zhang123huang, 2018-11-10:
CMake Error at /usr/local/share/catkin_simple/cmake/catkin_simple-extras.cmake:38 (find_package):
  By not providing "Findvoxblox.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "voxblox", but
  CMake did not find one.

  Could not find a package configuration file provided by "voxblox" with any
  of the following names:

    voxbloxConfig.cmake
    voxblox-config.cmake

////How to solve the problem????

ZacharyTaylor, 2018-11-11:
What command are you using to build the package is it `catkin build voxblox_ros`?

zhang123huang, 2018-11-12:
> What command are you using to build the package is it `catkin build voxblox_ros`?

Thank you very much for considering this question.I solved this problem by  installing  some extra dependencies.

chowkamlee81, 2018-11-30:
Kindly let us know what dependencies since iam also facing same issue

zhang123huang, 2018-12-02:
using the following cods

1 cd catkin_ws

2 rosdep install --from-paths src --ignore-src   

(you will find some dependencies name. 

  After that, you need install them one by one under src folder from GitHub. )