ethz-asl/voxblox #235: Freespace PointCloud Clearing Time

raghul9706, 2019-01-02:
Hello,
        As suggested in #156 and #154 , I implemented the Freespace pointcloud (containing points that are above certain depth ) . I placed an object in a open horizon in a simulated environment and removed it. Now the voxels created are getting cleared due to the Freespace Pointcloud behind after removing. The issue is the voxels get cleared much slower than that would have cleared if an obstacle is present behind. So it looks like, Time taken to clear a voxel with a ray casted from freespace point is much higher than that from a voxel of  obstacle behind (say 4 seconds and 1 second respectively). Regarding to an obstacle behind, the clearing rate is impressive. Are there any ideas on what the reason might be?

helenol, 2019-01-08:
Hmm... Haven't really experienced this issue before. Could you try playing with the weighting a bit and see if that affects it? `use_const_weight` to true in the params. Maybe try also with different integrators and see how the issue chnages? i.e., `fast` since that's the one we usually use.
Otherwise could you upload a sample bag with both the real + cleared pointcloud to try out that shows the effect?

raghul9706, 2019-01-08:
Hello Helen,
   Thanks a lot for the response.
Tried with fast integrator and the thing is it seems slower than merged. Also, didnt configure the exclusive params for Fast integrator (that might be the issue).
However in merged mode, setting _use_const_weight_ to true did not speeden it up. 
Once I get back to work, I will upload the bag file. Sorry for that :( 

**Some more params for reference**

- voxel_size : 0.10

- tsdf_voxels_per_side : 16

- max_ray_length_m : 5

- esdf_max_distance_m : 5

raghul9706, 2019-01-22:
Hey,
     Sorry to bother. Any update on this?

raghul9706, 2019-01-22:
Hey guys,
             Any update on this?

raghul9706, 2019-01-22:
Hello,
      Any updates? I am  currently working by constructing wall in the input pointcloud. It would be extremely helpful if freespace pointcloud does the job,pretty quicker.
Thank You.

rghl3, 2019-06-14:
Hi @helenol ,
       Any updates on this?

rghl3, 2019-07-03:
Hi, 
   Regarding the above issue, the rays from freespace points/ points beyond max_ray_length seem to get cleared when considering all points for clearing, by commenting out this [line](https://github.com/ethz-asl/voxblox/blob/9de0d2bc7e47383a79803d3fda447c27848529b4/voxblox/src/integrator/tsdf_integrator.cc#L400). But the voxels aren't being constructed for sparse obstacle points within the range. Is there any workaround? Thank You.
Regards, 
Raghul.

OPyshkin, 2022-03-15:
Hello. Can you please explain how you generate freespace pointcloud?