ethz-asl/voxblox #236: voxblox_ros launch failed

shadowver, 2019-01-11:
Thanks for the great work! I have followed every single step of the tutorials and successfully built the project. However,  when I ran roslaunch voxblox_ros basement_dateset.launch the error message indicated that "neither voxblox_ros nor basement_dateset.launch is not  a launch file". since I am new to the ROS, any help will be appreciated.

helenol, 2019-01-11:
Hi Joe,
You need to download some of the sample datasets, put them somewhere on your system, and open up the launch files and change the paths. :)

Hope that helps!
Helen

helenol, 2019-01-11:
Whoops additionally actually looks like the issue is that voxblox_ros isn't in the search path. Try doing `source  ~/catkin_ws/devel/setup.bash` and then `roslaunch voxblox_ros basement_dataset.launch`
If in doubt, `roscd voxblox_ros` should at least work to take you to the voxblox_ros folder, or ROS isn't sourced correctly.

shadowver, 2019-01-12:
Hi Helen,
Thank you for the reply. I will follow your second sugguestion as the error I encounterd occurred after I downloaded the dataset and path modification. I will let you know what is the result. ^.^

shadowver, 2019-01-14:
source the .bash file perfectly resolve the issue. Thanks a ton. =)

closed 2019-01-14 by shadowver