ethz-asl/voxblox #240: No ESDF with input from RealSense435

nitin5, 2019-02-25:
Greetings,
Thanks for open sourcing the code. 
I am trying to use ESDF output for reactive response from a platform. Sensor on board is Intel Realsense 435. I will like to run esdf_server node at atleast 4 Hz as mentioned in the paper of VoxBlox.
On launch of esdf_server node, output is empty mesh. All published topics have empty data structure. I have checked the rqt_graph, all relevant topics are well connected. Do esdf_server node expect any other input _e.g._ odometry _etc_. ?
Is there any list of cause of error which I shall check before diving into the code?
Rgds,
Nitin

mechaliomar, 2019-04-21:
Hello, I am also planning to do the same as you . Had you solved the problem. I see that the authors are no longer answering our questions !!!!

ZacharyTaylor, 2019-04-21:
The issue as you guessed is that Voxblox is not a full SLAM pipeline and only performs dense mapping. Because of this you need to provide it with an accurate 6-dof sensor pose before it will fuse in pointcloud measurements. By default this pose is taken from the ROS tf tree going from the frame "world" to the frame the pointcloud is given in.