ethz-asl/voxblox #244: Bad mapping results from basement dataset

bangyanz, 2019-03-07:
Hi

I was running the voxblox_ros basement_dataset on my computer with all default settings. After the processing is completed, it seems like what I got is a very bad map with huge empty spaces between small mapped areas. Is that I should expect? 
[image]

thank you.

ZacharyTaylor, 2019-03-08:
When you say defaults, do you mean without any parameters set, or using the ones in the basement_dataset launchfile (https://github.com/ethz-asl/voxblox/blob/master/voxblox_ros/launch/basement_dataset.launch)

With the launchfile you should be getting a much better reconstruction then the one above.

With pure defaults you will get quite a bad reconstruction for two reasons
1) the defaults are set for depth cameras and the dataset uses lidar (point density, noise model, max distance are all set to very wrong values)
2) the dataset has an odometry frame that exhibits a large amount of drift, and needs the optional ICP refinement enabled to compensate for it.

bangyanz, 2019-03-11:
Hi Zach, 

I doubled checked and I used the same launch file. 
[image]

What is worse, my cow and lady and all other datasets seem to have bad  results
/home/bangyan/Pictures/Screenshot from 2019-03-11 09-52-52.png

By 1, do you mean that my dataset yaml could be wrong? I git diffed my entire my project and there is no difference from the official version. 

By 2, I saw that icp is enabled but icp refine roll and pitch is not . Is that the optional ICP refinement you are talking about ?

Thank you.

ZacharyTaylor, 2019-03-11:
[image]

In my previous comment I was just making sure you were using roslaunch, and not rosrun as that would have loaded different settings.

I just checked out master of voxblox and ran it on the basement dataset. It seemed to work fine (see image). One thing you could try is to decompress the rosbag before running it as in the past I have found this sometimes causes issues.

Otherwise maybe paste the log you get when voxblox is running so I can try see whats going on.

bangyanz, 2019-03-12:
Hi, Zach 
@ZacharyTaylor I listened to your advise and decompressed the basement rosbag file (1.2GB after decompression). But the result is equally bad. 
[image]

I think I might have found out the problem: 
[image]
[image]

There is this error in the log. But I tripple checked and made sure my version is on master. I am running Geforce 1050 with i5 with Ubuntu 16.04 .

bangyanz, 2019-03-12:
I found out that it seems to take some time for the transform to kick in even though rosbag is already playing. I don't know if this error is the reason I have a bad map. 

ICP also show 0 successful update steps many times.

mfehr, 2019-03-12:
Hi @bangyanz
Might be a stupid question, but are you building in Release mode?

bangyanz, 2019-03-12:
@mfehr @ZacharyTaylor  
Looks like i made a stupid mistake and built in debug mode. 
Building in release mode does fix the problem.
Pardon my ignorance, but why does debug mode build generate a bad map ? 
I thought debug/release mode would normally only affect performance and not the quality of the result.

bangyanz, 2019-03-12:
Thank you for your help . I really appreciate it.

mfehr, 2019-03-12:
these things happen, often :)

It's just very slow, starts dropping data from the queues and probably the pose lookups fail due to delays, reconstruction is getting worse/sparse, ICP cannot align properly and makes mistakes, pointclouds are fused based on bad icp measurements and so on.

closed 2019-03-12 by mfehr

YJonmo, 2020-04-01:
> Hi, Zach
> @ZacharyTaylor I listened to your advise and decompressed the basement rosbag file (1.2GB after decompression). But the result is equally bad.
> [image]
>
> I think I might have found out the [...]

Hi there,
I am having exactly the same error message as you posted and it takes forever for the ros run the basement example.

I have Ubuntu 18 and installed using
sudo apt install ros-melodic-desktop-full

So I guess it is by default in the release mode?

[image]