ethz-asl/voxblox #246: ESDF map  update from TSDF

raghul9706, 2019-03-12:
Hi Team,
   I built the voxblox repo in release mode. When I run in **fast** mode, the tsdf map is updated when I remove the obstacle. But the corresponding region is not being updated correctly in Esdf map. Am I missing any configuration settings?
```Parameters``` 
- voxel_size : 0.10
- tsdf_voxels_per_side : 16
- use_tf_transforms : true
- max_ray_length_m : 10.0
- esdf_max_distance_m : 5.0
- esdf_default_distance_m : 5.0
- use_const_weight : false
- allow_clear : true
- voxel_carving_enabled : true

Rest all are default parameters.

**Before placing obstacle** 
[image]

**After Placing obstacle** 
[image]

**After removing obstacle**
[image]

Also another issue is when I rotate the robot_frame, the esdf pointcloud seems to be distinct for both the poses without merging.
[image]

The above issues persist even in merged mode too. 
Thank You.

fabianbl, 2019-04-17:
@raghul9706 can you check if this improves if in the esdf integrator config, you set `min_distance_m` lower than the default value? (I'm not sure if this parameter is exposed in case you are using `voxblox_ros`, but you can change it directly here for a quick check: https://github.com/ethz-asl/voxblox/blob/master/voxblox/include/voxblox/integrator/esdf_integrator.h#L46)

rghl3, 2019-06-14:
Hi @fabianbl ,
 Sorry for the late response. #259 seems to solve the issue for me with initial tests. Closing the issue. Thank You.
Regards,
Raghul

closed 2019-06-14 by rghl3