ethz-asl/voxblox #250: Reactive Collision Avoidance, based on ESDF

clmpng, 2019-03-21:
Hi,

I am using the voxblox node for planning and it works very well. But if I move objects within the planning area, the TSDF map reacts very slowly to that change. The ESDF is little faster but not much. I speeded up the ESDF generation by lowering the esdf_max_distance_m parameter. But if I visualize the esdf slice and set voxel_carving_enable to true, I always get a full esdf-layer - no matter how esdf_max_distance_m is set. How do that parameter relates to the esdf map, what distance does it influence?
Are there any other parameters for TSDF or ESDF generation that can be changed to improve reactive collision avoidance?

Kind regards,
Christian

Jaeyoung-Lim, 2019-03-22:
@clmpng The [mav_voxblox_planning](https://github.com/ethz-asl/mav_voxblox_planning) might be helpful for you

clmpng, 2019-03-22:
@Jaeyoung-Lim Thank you, but I'm using that package already - it works good, only the collision avoidance reacts very slowly.

Jaeyoung-Lim, 2019-03-22:
@clmpng By reading again, by moving the object in the scene, are you trying to use Voxblox for dynamic scenes? Voxblox assumes you have a static environment

clmpng, 2019-03-22:
@Jaeyoung-Lim Yes, I try to used it for dynamic environments. It works well with a low voxel resolution, since the esdf map is generated very quickly in that case. Now, I'm trying to change some other parameters to make it work with a higher resolution. For that reason, I was wondering about the impact of the esdf_max_distance_m parameter.

raghul9706, 2019-03-25:
Hi @clmpng, 
       Say your esdf_max_distance_m  is 5m. Then all voxels 5m away from an obstacle would have their distances as 5m. Try reducing the **max_ray_length_m**. Hope that's the parameter to meet your above expectations. This parameter limits the ray cast to the specified distance.

mechaliomar, 2019-04-21:
hello guys, I'm new here. I have been following the tutorial step by step in https://github.com/ethz-asl/mav_voxblox_planning , but unfortunately I can't find out how to solve a problem that I encountered.
I cant do the following (and start up rviz (rviz). In rviz, select Panels -> Add New Panel and select Planning Panel)

[image]
[image]

ZacharyTaylor, 2019-04-21:
Do you have `source ~/catkin_ws/devel/setup.bash` as part of your bash.rc? If you don't you need to run this command in the terminal you use to open rviz, otherwise it won't know which directories to search for the plugin.

mechaliomar, 2019-04-22:
@ZacharyTaylor Thanks for your answer. I followed exactly that by typing:

cd ~/catkin_ws_path/src/
source ~/catkin_ws_path/devel/setup.bash

And I opened other terminal on which I typed 
roscore

then on the first one I typed 
rviz

[image]

SkyRiderMike, 2020-05-22:
Could you give me any advice about  using Voxblox to generate a 2.5D map ?