ethz-asl/voxblox #258: How to use my own camera to build the map instead of using the dataset?

mechaliomar, 2019-04-28:
Hello,
I would like to know step by step how to  use m own camera to build the map instead of using :
cow_and_lady_dataset.launch or other examples.

BEST REGARDS

ZacharyTaylor, 2019-04-28:
To work Voxblox needs a system to provide it with three pieces of of information

1) A 3D pointcloud of the environment, this will be remapped to voxblox's `pointcloud` topic. There are quite a few ways to get this, we have had success using
- Depth cameras such as the realsense or Kinetic
- Stereo cameras such as the VI-sensor, combined with a dense stereo pipeline (see KITTI-dataset example).
- 3D lidars such as made by Velodyne or Ouster

2) An estimate for the 6 degree of freedom pose of the system.  This can be given using the ros `tf` tree (default option). In this case voxblox will look for a transform from the `world` frame to the frame the above point-cloud is given in. A seperate ros node is required to give this transform. Here there are also many options, we have previously used
- Vicon / optitrack systems
- Visual slam solutions such as ORB-SLAM
- Visual-inertial solutions such as ROVIOLI, ROVIO or VINS-Mono.
- Lidar ICP solutions

3) The transformation from the output of 2) to the frame the pointcloud in 1) is given in. If the same sensor is used for both this can just take the default value of identity.

mechaliomar, 2019-04-29:
>    1. An estimate for the 6 degree of freedom pose of the system. This
>    1. The transformation from the output of 2) to the frame the

Thanks a lot for your precise answer.

In our lab we have a visual-inertial sensor  (it contains bothe camera and
IMU ), we are preparing to use it for the pointcloud and for the DOF
estomation.

But where could we find the appropriate '. launch' file which will aquire
data from the sensor and also run the needed nodes to create the map? (I
mean in your open source code , you provide many launch file to be used
with different dataset in voxblox_ros->launch, but how about the real time
acquisition from the real sensor). Hope you understand me

BEST REGARDS

ZacharyTaylor, 2019-04-29:
You will have to provide the node that acquires the data from the sensor and publishes it as a ros topic yourself. After that you can just take one of the launchfiles, add your sensor node and remove the parts that play the bag. Alternatively to just run with all default parameters you can just add the following to your own launchfile 

```
  <node name="voxblox_node" pkg="voxblox_ros" type="tsdf_server" output="screen">
    <remap from="pointcloud" to="YOUR/POINTCLOUD/TOPIC/HERE"/>
  </node>
```

mechaliomar, 2019-04-30:
ok, thank you for your response . I really appreciate that.

mechaliomar, 2019-04-30:
Hi, thanks for your response,I really appreciate that

mechaliomar, 2019-04-30:
here is my launch file, but it's not working, 

<launch>

  <arg name="play_bag" default="true" />
  <arg name="bag_file" default="/home/omar/Downloads/data.bag"/>
  <arg name="voxel_size" default="0.05"/>

  

  <node name="player" pkg="rosbag" type="play" output="screen" args="-r 1.0 --clock $(arg bag_file)" if="$(arg play_bag)"/>

   <node name="voxblox_node" pkg="voxblox_ros" type="esdf_server" output="screen" args="-alsologtostderr" clear_params="true">

    <remap from="pointcloud" to="/camera/depth_registered/points"/>

    <param name="tsdf_voxel_size" value="$(arg voxel_size)" />
    <param name="tsdf_voxels_per_side" value="16" />
    <param name="voxel_carving_enabled" value="true" />
    <param name="color_mode" value="color" />
    <param name="use_tf_transforms" value="false" />
    <param name="update_mesh_every_n_sec" value="1.0" />
    <param name="min_time_between_msgs_sec" value="0.0" />
    <param name="method" value="fast" />
    <param name="use_const_weight" value="false" />
    <param name="allow_clear" value="true" />
    <param name="verbose" value="true" />
    <remap from="transform" to="/kinect/vrpn_client/estimated_transform" />
    <rosparam file="$(find voxblox_ros)/cfg/cow_and_lady.yaml"/>
    <param name="mesh_filename" value="$(find voxblox_ros)/mesh_results/$(anon cow).ply" />
  </node>

</launch>

mechaliomar, 2019-04-30:
<launch>
  <arg name="play_bag" default="true" />
  <arg name="voxel_size" default="0.05"/>
   <node name="voxblox_node" pkg="voxblox_ros" type="esdf_server" output="screen" args="-alsologtostderr" clear_params="true">

    <remap from="pointcloud" to="/camera/depth_registered/points"/>

    <param name="tsdf_voxel_size" value="$(arg voxel_size)" />
    <param name="tsdf_voxels_per_side" value="16" />
    <param name="voxel_carving_enabled" value="true" />
    <param name="color_mode" value="color" />
    <param name="use_tf_transforms" value="false" />
    <param name="update_mesh_every_n_sec" value="1.0" />
    <param name="min_time_between_msgs_sec" value="0.0" />
    <param name="method" value="fast" />
    <param name="use_const_weight" value="false" />
    <param name="allow_clear" value="true" />
    <param name="verbose" value="true" />
    <remap from="transform" to="/kinect/vrpn_client/estimated_transform" />
    <rosparam file="$(find voxblox_ros)/cfg/cow_and_lady.yaml"/>
    <param name="mesh_filename" value="$(find voxblox_ros)/mesh_results/$(anon cow).ply" />
  </node>

</launch>

mechaliomar, 2019-04-30:
Here is my launch file , but it's not working

[myfile.txt](https://github.com/ethz-asl/voxblox/files/3131190/myfile.txt)

helenol, 2019-04-30:
```
    <param name="use_tf_transforms" value="true" />
```
And make sure you have a transformation between the world/odometry frame and the pointcloud frame which you need to get using a state estimator.

mechaliomar, 2019-05-08:
@helon I think I dont neet to use        because  I'm using the same sensor for point cloud  and IMU data acquisition.
@ZacharyTaylor  I created a node to acquire data from the sensore and publish the pointcloud  to a topic and I also remapped it to voxblox as you said:

    <node name="mynteye_wrapper_node" pkg="mynt_eye_ros_wrapper" type="mynteye_wrapper_node" output="screen" respawn="true" respawn_delay="10">
  </node>
   <node name="voxblox_node" pkg="voxblox_ros" type="esdf_server" output="screen" args="-alsologtostderr" clear_params="true">
    <remap from="pointcloud" to="/points/data_raw"/>

But I think, I still need some modification for in launch file, for example I don't know what do the following lines mean:

    <remap from="transform" to="/kinect/vrpn_client/estimated_transform" />
    <rosparam file="$(find voxblox_ros)/cfg/cow_and_lady.yaml"/>
    <param name="mesh_filename" value="$(find voxblox_ros)/mesh_results/$(anon cow).ply" />

Abother question: How to provide the IMU data to the voxblox_node because it's not subscribing to the IMU topic as follow:
omar@omar:~$ rostopic info /imu/data_raw
Type: sensor_msgs/Imu

Publishers: 
 * /mynteye_wrapper_node (http://localhost:39847/)

Subscribers: None

ZacharyTaylor, 2019-05-09:
Voxblox doesn't use IMU data and so can't subscribe to it. The three parameters you are unsure of are not needed for your setup. Currently you are still missing a pose estimate and so nothing will work until one is provided.

mechaliomar, 2019-05-09:
@ZacharyTaylor 
*** ok I understand. SO in the next step I'll try to use ROVIO for the 6 DOF pose estimation as it is illustrated in the following image:
[image]

*** I have a question:
While I finish the pose estimation using ROVIO , I need another node which  read the 6 DOF pose estimation from  ROVIO and publish it to a topic, then the voxblox_node subscribe to that topic and read the 6 DOF pose !!
IS THAT CORRECT?

@ZacharyTaylor  DO YOU KNOW   Mr Mohamed Bouri? I think he works in your university

eyildiz-ugoe, 2019-06-11:
Dear authors,

In order to try our the code with our own setup, a convenient way of making voxblox work is required. People have different environments and thus they may like to see the system performing on objects different than cows and basements. It's safe to assume they would have a point cloud provider, as well as a working ROS system. 

Given this, could you perhaps provide a proper guideline or at least the node with your own parameters (with a real camera, not bag files), which we can then modify and test the code in our environments? Because many people are after this kind of setup, not cow/basement stuff. We want to test if the mesh creation out of our own objects is successful or not.

> 1. An estimate for the 6 degree of freedom pose of the system.  This can be given using the ros `tf` tree (default option). In this case voxblox will look for a transform from the `world` frame to [...]

What if we only have a camera and not a robot? Can we artifically create a urdf model and then pretend as if there is a robot base which is placed at (X,Y,Z), and then use the TF? What if we simply want to create a mesh out of a point cloud  without having a drone or a robot?

hyclover, 2021-11-16:
@mechaliomar ， Hello, I use orb_slam2 to provide pose estimate. But I still encounter the problem like yours. Have you solved this with ROVIO?

mechaliomar, 2021-11-16:
@hyclover. It has been a long time. I canceled that work. Good luck