ethz-asl/voxblox #260: what's the difference between maplab and voxblox?

mechaliomar, 2019-05-12:
maplab : https://github.com/ethz-asl/maplab

mfehr, 2019-05-12:
after reading our papers again and a lot of discussions we came up with the following list of differences:
[image]

Sources:
 - https://en.wikipedia.org/wiki/Sheep
 - https://en.wikipedia.org/wiki/Fox
 - https://arxiv.org/abs/1711.10250
 - https://helenol.github.io/publications/iros_2017_voxblox.pdf

mechaliomar, 2019-05-14:
@mfehr Marius it's funny but so meaningful. Thanks

closed 2019-05-14 by mfehr

HongLouyemeng, 2022-01-06:
> 

> after reading our papers again and a lot of discussions we came up with the following list of differences: [image]
> 
> Sources:
> 
> * https://en.wikipedia.org/wiki/Sheep
> * https://en.wikipedia.org/wiki/Fox
> * https://arxiv.org/abs/1711.10250
> * https://helenol.github.io/publications/iros_2017_voxblox.pdf

This reply is very beautiful!