ethz-asl/voxblox #262: Voxblox and Long Term Mapping

dario-ljubic, 2019-05-16:
Hello, 
I've been working with Voxblox for a while now and I just want to thank you for sharing this amazing project.

I have a question regarding Voxblox and its' ability for long term mapping.
After experimenting with large maps, the algorithm, as expected, slows down at a certain map size and also fills up my RAM.
Have you thought about long term mapping in a sense that parts of the map would be kept on a hard drive and would be loaded again as a robot revisits the previously mapped area?

Do you have a plan of implementing something like that and do you think it is feasible or maybe have the advice of how would someone go about doing it (especially the part of loading TSDF and "stitching" it to the local map)?

Thank you in advance and keep up the good work!

helenol, 2019-05-16:
Hey! Yes we've thought about this a lot in the design. You should actually be able to do this largely with components that are already there, though there might be some elbow grease required with implementing some features in the serialization/deserialization, and definitely in the bookkeeping.

The nice property of block-hashing is that all blocks are treated independently in memory. This means that you can arbitrarily drop blocks from memory whenever you like. You can take a look at how the "remove_distant_blocks" is implemented at the Layer level: https://github.com/ethz-asl/voxblox/blob/master/voxblox/include/voxblox/core/layer.h#L169

I think the general idea would be to, instead of deleting the blocks, just selectively serialize them. You'd then have to keep track of what block hash-ids are serialized, and then before "allocating" them again in the layer level, pull them from harddrive instead.

If you get this up and running, we'd love to have a pull request for this feature! :) I think it'd help a lot of people.

Also for a consistent voxbloxing framework in the presence of loop-closures etc, check out c-blox: https://github.com/ethz-asl/c-blox and its corresponding paper: https://arxiv.org/abs/1710.07242
As a preview, we'll have an extension of that work with its own pose-graph back-end and optimization coming in the next few months.

ZacharyTaylor, 2019-05-16:
In addition to the above most of the runtime of the TSDF part of voxblox is
independent of the map size. The only exceptions are some of the outputs,
such as outputting the map as a pointcloud and the getAllUpdatedVoxels
function that loops through all map blocks, however it would be a simple
change to push blocks onto a queue when they are updated, it's just that
when we profiled the code this was never a bottleneck.

Meshing is also independent of the map size, but displaying it isn't and so
RViz tends to become the bottleneck and start crashing at a few million
triangles.

dario-ljubic, 2019-05-21:
Thank you both very much!

Closing the issue...

closed 2019-05-21 by dario-ljubic