ethz-asl/voxblox #263: ESDF from scratch every iteration/"forget-factor"

tsijs, 2019-05-16:
Hi all, 

I am using voxblox for esdf generation for planning purposes. In a static environment, this works very well. Now, I realize voxblox is not made for dynamic environments, but I was hoping there is a way to compute the ESDF as if it was the first iteration for every iteration or if there is a parameter I can adjust to increase the "forget-factor" per iteration s.t. I can use voxblox in a dynamic environment. 
Many thanks in advance!

helenol, 2019-05-20:
@tsijs, are you having specific issues with dynamic obstacles? If the incremental and the batch ESDF generation are working differently, then we have a bug. :) (We recently fixed this issue: https://github.com/ethz-asl/voxblox/pull/256 which was related to this exact problem, so pull master if you hadn't in a while).
The ESDF is just made from the TSDF, so you could play with the weighting in the TSDF to remove dynamic obstacles faster.
In our experience, people moving and so on get cleared in 4-5 scans depending on settings.

tsijs, 2019-05-30:
Thank you for the fast response, I do have issues with dynamic obstacles indeed. I tried playing with the parameters in the TSDF, which made it a little better, but it still takes +-2 seconds before an obstacle that moved disappears in the environment representation, which is too slow for my application and also slpwer than what you described I think. I use an update rate of 10Hz and the "fast" integration setting. Could you maybe give me some pointers to the parameters that should be readjusted to get better performance?

Magnusgaertner, 2019-06-13:
Hi @tsijs & @helenol ,
I am also experimenting with non static scenes and therefore set allow_clear=true, use_const_weight=true, max_weight=1. Especially max_wheight=<some low value>

the full parameters (that differ from default) are:

```
<node name="voxblox_node" pkg="voxblox_ros" type="esdf_server" output="screen" args="-alsologtostderr" clear_params="true">
            <remap from="pointcloud" to="/sampled/points"/>
            <remap from="voxblox_node/esdf_map_out" to="esdf_map" />
            <param name="tsdf_voxel_size" value="$(arg voxel_size)" />
            <param name="tsdf_voxels_per_side" value="$(arg voxels_per_side)" />
            <param name="publish_esdf_map" value="true" />
            <param name="publish_pointclouds" value="true" />
            <param name="use_tf_transforms" value="true" />
            <param name="update_mesh_every_n_sec" value="0.2" />
            <param name="clear_sphere_for_planning" value="false" />
            <param name="world_frame" value="$(arg world_frame)" />
            <param name="verbose" value="false"/>
            <param name="method" value="fast" /> <!--fast, merged, simple -->
            <param name="use_const_weight" value = "true"/><!--for laserscanners -->
            <param name="esdf_min_diff_m" value="0.1"/>
            <param name="max_weight" value="1" />
            <param name="allow_clear" value="true"/>
        </node>
```

But I am experiencing some artifacts in the esdf generated from moving artifacts. The video shows these artifacts, @helenol any guess what could cause this behaviour?

[Video](https://drive.google.com/file/d/1EuwH8oExgFHrhEJqcx4Nw94N6nP7zknv/view?usp=sharing)