ethz-asl/voxblox #264: Multiple pointclouds as source for ESDF

tsijs, 2019-05-16:
Hi again, 

Another question:
It seems that voxblox currently only supports a single source topic for the ESDF generation. 
Would it be possible to extend this to multiple source topics and concatenate the pointclouds s.t. multiple cameras can be used as a sensor? 

I was thinking about implementing this, but I am uncertain what the best way to deal with this would be. 

Any thought on this are welcome.

ZacharyTaylor, 2019-05-16:
If you are using a ros tf tree for your transform inputs this is actually already implemented, just map all of your input pointclouds to the same topic. The frame they are given in will allow voxblox to associate them with the correct location. You will probably also need to set the min time between pointcloud integrations to zero.

tsijs, 2019-05-30:
This works indeed well enough for my application. Thank you!

closed 2019-05-30 by tsijs