ethz-asl/voxblox #272: Only ESDF generation

tsijs, 2019-05-30:
I want to strip everything from the voxblox_ros node that is not necessary for the ESDF computation. I only want to compute the ESDF and send it and do this as fast as possible, since I want to use it for moving obstacles. 

Is there a way to do this with just setting parameters or should I carve out some code manually?

Looking at the paper, I think it should at least be possible to not use any mesh generation/updating, since that seems to be independent of the ESDF.

ZacharyTaylor, 2019-07-07:
The TSDF is required by the ESDF so would need to be kept. You can disable the mesh generation and some of the outputs. The only place I could see you getting a notable speed gain over the default implementation is if you were to remove the color information and its integration from the TSDF blocks.

tsijs, 2019-07-09:
Thanks for the information. However, due to unrelated issues to this I stopped using the voxblox package for ESDF generation for dynamic environments. I will close this issue. Sorry for not closing it earlier:$

closed 2019-07-09 by tsijs