ethz-asl/voxblox #280: A question of TSDF/ESDF definition

im-Kitsch, 2019-07-04:
Hi, I have a basic problem of the distance definition of TSDF/ESDF.  The distance is defined as from this point to nearest surface. But this distance, from point to surface, is defined as general geometriy definition, i.e. vertical distance or we don't assume that the surface is infinite by calculating the surface?

Like following 2D case, we assume Point C and Point A are respectively (0, 0) and (3, 4), and Point C is corner point of surface s_1 and s_2.

So the distance of point A in TSDF/ESDF are just 5 or 3, (directly form point A to C or from A to surface s_1 )

```
      
                                               Point A                        

surface s_1
--------------------------------- C
                                 |
                                 |      surface s_2
                                 |
                                 |
```

ZacharyTaylor, 2019-07-07:
We only measure to observed points. For the ESDF the distance in the example is 5. The TSDF uses a projective distance measure and so will give a value that is greater than or equal to 5.

resplendent-star, 2020-05-21:
> We only measure to observed points. For the ESDF the distance in the example is 5. The TSDF uses a projective distance measure and so will give a value that is greater than or equal to 5.

hello，do  you  mean  the  TSDF distance of  point A，is  let  a  ray  light  come  from  camera optical  center  to  point  A，and  obviously   this ray  will have  an  intersection   【point  I】in  a  surface，and  the  TSDF distance is  Euclidean distance  between  I and  A

im-Kitsch, 2020-05-24:
Thanks for the help.  :)

closed 2020-05-24 by im-Kitsch