ethz-asl/voxblox #284: Mesh generation using voxblox

pavan4, 2019-07-18:
Hello, 
I have a custom dataset in the [TUM dataset](https://vision.in.tum.de/data/datasets/rgbd-dataset/file_formats) format captured using a depth sensor. I looked at the cow and lady dataset rosbag and created a rosbag in the exact format and ran voxblox on it. I am able to run voxblox, integrate and generate mesh **but** visually the mesh doesn't look correct. 

Below is generated mesh - 

[image]

A sample video frame - 

[image]

I have used the same parameters as cow and lady dataset for TSDFs integration. 

`T_B_C`  and `T_B_D` were set to identity matrix. 

Voxblox is receiving camera poses, point clouds and the integration is being performed but the mesh is incorrect as you can see.

I would like to ask if I am missing any obvious parameter or any other option that seems to be missing?

mfehr, 2019-07-19:
Hi @pavan4 

Most likely your source of transformations (World to Base/Camera/Depth) is still wrong. Are you using the tf tree or the transform topic as source? If you use the former, check the tf tree of your bag and the frame names you pass in as world/sensor frame. If it's the latter the most likely thing that is wrong is simply the transformation itself. How I usually debug these things is to run the bag and visualize the point cloud in global coordinates (if you have tf tree that should automatically happen in rviz, for the stamped transform you need to transform it yourself or hack it into voxblox). If you visualize that way you should see pretty easily whether the point cloud makes sense and reflects the trajectory/viewpoints of the recording.

pavan4, 2019-07-19:
Hi @mfehr, thank you for this.  

`use_tf_transforms` is set to false using `<param name="use_tf_transforms" value="false" />`. So I am sure I am using the `transform` topic. 

I am using the camera transformation that was generated by the camera and create the TransformStamped using the transform values of `tx ty tx qx qy qz qw` from the same TUM dataset format. I am not sure if I have to modify the transformation into any format specific to voxblox. I still haven't figured out the where the issue is. I will try the visualization you suggested but if the transformation is incorrect then I am not sure how to check that. 

I also wanted to mention that I used [surfelmeshing](https://github.com/puzzlepaint/surfelmeshing) with the same transform and it worked without any issue.

mfehr, 2019-07-22:
Yes, I would try the visualization to debug this, could be that surfelmeshing expects a different direction of the transformation or is in a different frame/convention. Check whether it needs to be T_W_B (transforms points from base to world) or T_B_W (transforms points from world to base). The base frame should be the local sensor frame the point clouds are expressed in, and of course the point clouds should be in that frame when passing to voxblox.

pavan4, 2019-07-22:
@mfehr, thanks for the tip on visualization. I was able to fix and generate the mesh using voxblox.

The issue was - depthmap needed to be resized to rgb image dimensions with `intepolation=cv2.INTER_NEAREST` . If I just used plain `cv2.resize`, the noise introduced made the integration generate mesh which was incorrect.

closed 2019-07-22 by pavan4