ethz-asl/voxblox #290: Voxblox standalone library use

ltstein, 2019-07-31:
Is there any documentation on using voxblox outside of as a ROS node? Such as integrated with a standalone C++ application?

mfehr, 2019-07-31:
Hi @ltstein 
The code API is documented [here](https://voxblox.readthedocs.io/en/latest/api/library_root.html). We deliberately split voxblox into the voxblox package without ROS and voxblox_ros and we have been using it independenty from the ROS nodes in other projects such as maplab. The API should be pretty straight-forward to use if you use the ROS nodes as a code example, otherwise you can also check out how we do it in maplab ([integration](https://github.com/ethz-asl/maplab/blob/pre_release_public/july-2018/interfaces/voxblox_interface/src/integration.cc#L108), [setting up the map](https://github.com/ethz-asl/maplab/blob/pre_release_public/july-2018/console-plugins/dense-reconstruction-plugin/src/dense-reconstruction-plugin.cc#L304), [setting up the integrator](https://github.com/ethz-asl/maplab/blob/pre_release_public/july-2018/interfaces/voxblox_interface/src/integration.cc#L170)), although the public version is using an outdated version and there might be some changes.

ltstein, 2019-07-31:
Thank you for the quick response. I got the library working with ROS and was starting to look at the ROS nodes as an example and I just wanted to make sure I wasn't missing anything.

closed 2019-07-31 by ltstein

apache8080, 2020-10-21:
For building the standalone library, it seems like I still need ROS. Is there a way to avoid this?