ethz-asl/voxblox #291: Input pointcloud queue getting too long!

wangyuanbiubiubiu, 2019-08-01:
When I run the program, I get this error. My point cloud is released once every 10HZ or so, and the scene is similar to KITTI. The vehicle is driving on the road, with a single purpose. The following are my parameters. May I ask which parameters I should change
[image]
[image]

wangyuanbiubiubiu, 2019-08-01:
And then I meet this issue
[image]

Super-Iron-Man, 2020-07-29:
hi, I encountered the same problem as you, have you found a solution?

Super-Iron-Man, 2020-07-29:
@wangyuanbiubiubiu Hi,I encountered the same problem as you, have you found a solution?

chris-aeviator, 2020-12-04:
Running into this when using ros_tango_streamer (android app that runs with a tango/tof sensor)

chris-aeviator, 2020-12-04:
I can get rid of the error by raising `kMaxQueueSize` (https://github.com/ethz-asl/voxblox/blob/66a6b1cf17f9fa9cd3222e450b09f85e61d17d93/voxblox_ros/src/tsdf_server.cc#L329) to something ridiculously high (e.g. from 10 to 100000) but still no tsdf or mesh is produced, no further error is raised

ZWT006, 2022-11-15:
@wangyuanbiubiubiu I having seen the same problem when I using my own ros.bag. The divice is D435, the pointcloud is 6Hz. And there are also some wrang:`No match found for transform timestamp`
The launch parameters as follows:
```
 <arg name="play_bag" default="false" />
  <arg name="bag_file" default="/media/zwt/UbuntuFiles/pcl_data/rosbagdatas/A1_TD_v1.bag"/>
  <arg name="voxel_size" default="0.20"/>

  <!-- Download the dataset here: https://projects.asl.ethz.ch/datasets/doku.php?id=iros2017 -->

  <node name="voxblox_node" pkg="voxblox_ros" type="tsdf_server" output="screen" args="-alsologtostderr" clear_params="true">
    <remap from="pointcloud" to="/d435/depth/color/points"/>
    <remap from="voxblox_node/esdf_map_out" to="esdf_map" />
    <param name="tsdf_voxel_size" value="$(arg voxel_size)" />
    <param name="tsdf_voxels_per_side" value="16" />
    <param name="publish_esdf_map" value="false" />
    <param name="publish_pointclouds" value="true" />
    <param name="voxel_carving_enabled" value="true" />
    <param name="color_mode" value="color" />
    <param name="use_tf_transforms" value="true" />
    <param name="update_mesh_every_n_sec" value="0.5" />
    <param name="min_time_between_msgs_sec" value="0.0" />
    <param name="clear_sphere_for_planning" value="true" />
    <param name="method" value="fast" />
    <param name="use_const_weight" value="false" />
    <param name="allow_clear" value="true" />
    <param name="verbose" value="false" />
    <remap from="transform" to="/vio_transform" />
    <param name="world_frame" value="t265_odom_frame" />
    <param name="sensor_frame" value="d435_depth_optical_frame" />
    <rosparam file="$(find botmap)/config/t265_and_d400.yaml"/>
    <param name="mesh_filename" value="$(find voxblox_ros)/mesh_results/$(anon A1).ply" />
    <param name="pointcloud_queue_size" value="3" />
  </node>
```

Mr-brillianter, 2022-11-23:
the same.

bqm1111, 2025-03-05:
Anyone found a solution for this problem?

ZWT006, 2025-03-06:
> Anyone found a solution for this problem?

It seems like the timestampes of rosbag's topics are different. So, when we play our rosbag, we can try to use the `rosbag ... --clock` parameter to enable simulation time in rosbag and set the ros master as `rosparem set /use_sim_time true` to enable simulation time of master.

bqm1111, 2025-03-20:
I didnt run using rosbag but I ran directly on hardware. The problem still persist. Did you encounter the same problem on hardware?

> It seems like the timestampes of rosbag's topics are different. So, when we play our rosbag, we can try to use the `rosbag ... --clock` parameter to enable simulation time in rosbag and set the ros [...]

ChanJoon, 2025-05-14:
If there is no TF between those frames, the pointcloud queue will always be empty.
So, check the TF tree and if it's difficult to handle, just comment out that part. Then it should work.