ethz-asl/voxblox #292: Getting the mesh via ROS

eyildiz-ugoe, 2019-08-06:
As it is stated in the docs, upon a service call, a mesh is saved under `/mesh_results`/. I would like to actually receive this mesh with my node, which will make the service call to Voxblox. At the moment Voxblock has a fixed path for saving it, so my questions are:

1. How can I receive the mesh upon a service call with my node? Can I, for instance, change the directory to save, or even better, can I direct the mesh to my node?

2.  In order to make this work, I need to have exactly the same messages as` Voxblox_ros` has, that are, `Mesh.msg` and `MeshBlock.msg` found under `/msg`. Do I need something else? Or can I simply copy-paste these messages into my own node's `/msg` and make the necessary `CMakeLists.txt` changes and that's it? 

Thanks in advance.

ZacharyTaylor, 2019-08-17:
Voxblox offers two ways of generating and sending meshes over ROS.

1) the "mesh" topic that sends messages of type: voxblox_msgs::MeshBlock
This topic ouputs blocks of mesh at a fixed update rate that have changed since the function was last called. It has the advantage of being fairly low bandwidth, but you have to do the bookkeeping on the mesh blocks and combining them on your side. See the voxblox_rviz_plugin node for an example of a node that does this receiving and processing.

2) the "mesh_pcl" topic that sends messages of type pcl_msgs::PolygonMesh
Only appears if output_mesh_as_pcl_mesh is true, outputs any mesh generated by the generate_mesh service. Potentially more in line with what you are looking for.

ToniRV, 2019-09-01:
Hi @ZacharyTaylor is  there any performance difference between using one or the other?

oleg-alexandrov, 2019-10-16:
The variables output_mesh_as_pcl_mesh and output_mesh_as_pcl_mesh don't show up in the code, only in the doc (if one searches the github repository). In contrast, the variable update_mesh_every_n_sec shows up in the code and is parsed.

What this likely means is that the documentation is outdated, I mean the page at: https://github.com/ethz-asl/voxblox/blob/b6cfd0999de6f8c81a351c81b6f1196c463b5d31/docs/pages/The-Voxblox-Node.rst

It would be nice to update it.