ethz-asl/voxblox #293: Ply writer doing something weird

ToniRV, 2019-09-01:
I am seeing a very weird behaviour when saving a mesh to a ply using the `generate_mesh` rosservice.
When I visualize the mesh in RVIZ, it is clear that the mesh in positive y, and z is up.
Yet when I write the mesh I see the values of y being the ones of z, and the ones of z being the ones of -y.

1. Here the RVIZ visualization:
[image]

2. Here the mesh loaded in another visualizer:
[image]

3. Here a sample of the ply file:

[image]

helenol, 2019-10-17:
Are you sure you are visualizing the mesh in the same frame that it is created in? I.e., the "world_frame" frame ID? Because the ply writer really does not do anything special and I really do not see how this would happen:
https://github.com/ethz-asl/voxblox/blob/master/voxblox/include/voxblox/io/ply_writer.h#L72-L98

ToniRV, 2020-03-14:
Thanks @helenol seems like it was just an intricate combination of frames of reference.

closed 2020-03-14 by ToniRV