ethz-asl/voxblox #298: Potential inaccurate documentation issue

oleg-alexandrov, 2019-10-16:
The documentation at 

https://voxblox.readthedocs.io/en/latest/pages/Running-Voxblox.html

says that /voxblox_node/mesh should publish a MarkerArray. Instead, it publishes the type voxblox_msgs/Mesh

Is there a mistake somewhere?

Speaking more generally, I have been trying to run the  cow and lady dataset, with no luck. Rviz does not display any mesh or point cloud whenever I subscribe to any published topic when I launch this dataset (but I can see the topics with rostopic list).

Any thoughts? Thanks.

alexmillane, 2019-10-16:
Usually this occurs because the map is built, and the mesh published, in a different frame than what you have in the Rviz parameter "Fixed Frame" under "Global options". I believe for "cow and lady" the frame should be set to `map`.

oleg-alexandrov, 2019-10-16:
Thank you. The frame is actually "world", rather than "map". The documentation got it right. What it got wrong is to say that the topic of /voxblox_node/mesh is MarkerArray. It is in fact voxblox_msgs/Mesh. I created a pull request with a fix. (Surely it is a tiny thing, just confusing for new users.)

oleg-alexandrov, 2019-10-18:
My documentation fix got accepted which fixed this, so I am closing it. But the update did not make it yet from 

https://github.com/ethz-asl/voxblox/blob/master/docs/pages/Running-Voxblox.rst

to the generated page at:

https://voxblox.readthedocs.io/en/latest/pages/Running-Voxblox.html

closed 2019-10-18 by oleg-alexandrov