ethz-asl/voxblox #304: How to use voxblox for mapping with MSCKF_VIO?

wntun, 2019-11-07:
Can anyone please guide me how to convert from SLAM odometry value to transform stamped data for the transform in voxblox? 

With MSCKF_VIO, odometry values are obtained and converted them to transform stamped using 
 tf::pointMsgToEigen for position and tf::quaternionMsgToEigen for orientation.

I tested with euroc dataset. If I use the provided value /vicon/firefly_sbx/firefly_sbx from the rosbag, it's no problem. With the conversion, I got the error given below. 

[image]

Any kind of help would be appreciated.

oleg-alexandrov, 2019-11-08:
As I see it, in the first screen shot, it gets a quaternion of zero length. So something got lost in translation. It is also complaining about not being able to find a timestamp, which is another bookkeeping issue.

wntun, 2020-08-06:
I'm still having troubles. This time, I wonder if the errors occurred because of conversion method from odometry to transformStamped. I'm modifying the method from transform_to_tf (transform_tools). Please help me if anyone knows how to correct it.

`void OdomToTransform::odomCallback(const nav_msgs::OdometryConstPtr& msg) {
          // Converting and storing the transform
        	odometry_.header = msg->header;
        	odometry_.child_frame_id = msg->child_frame_id;
        	odometry_.pose = msg->pose;
        	odometry_.twist = msg->twist;
          //tf::transformMsgToKindr(msg->transform, &transform_.transformation);
      }
    
    void OdomToTransform::publishTransformStamped(const ros::TimerEvent& event) {
    	// odom -> transformStamped (point)
    	geometry_msgs::TransformStamped transform;
    	transform.header = odometry_.header;
    	transform.child_frame_id = odometry_.child_frame_id;
    	transform.transform.translation.x = odometry_.pose.pose.position.x;
    	transform.transform.translation.y = odometry_.pose.pose.position.y;
    	transform.transform.translation.z = odometry_.pose.pose.position.z;
    	transform.transform.rotation = odometry_.pose.pose.orientation;
    
    	transform_pub_.publish(transform); 
    }`

alexmillane, 2020-08-06:
The problem indicated in your first error is that voxblox is receiving an unnormalized quaternion (norm 0). It's likely that it's receiving a quaternion that is all zeros.

wntun, 2020-08-06:
@alexmillane I thought the conversion method was wrong. I just added a condition to check for quaternion that is all zeros. It's working now. Thanks alot! 
[image]

alexmillane, 2020-08-06:
That great! Glad I could help :)

closed 2020-08-06 by alexmillane