ethz-asl/voxblox #306: No visualization on rviz

aaravrav142, 2019-12-05:
Hi, 

I correctly build the package and tried to run the cow and lady dataset using 
`roslaunch voxblox_ros cow_and_lady_dataset.launch`

On launching rviz and keeping fixed frame as '`world`' and using the voxblox plugin `/voxblox_node/mesh`, nothing displays on the rviz screen. On the console however everything seems to run fine. 
How can I solve this issue. Any help would be useful. Thanks

XanxusCrypto, 2020-01-09:
I met the same problem. Waiting for some help. Thanks for sharing.

nachovizzo, 2020-02-04:
That's because you didn't source the devel space before launching rviz most likely
Just `source ~/catkin_ws/devel/setup.bash` and then launch `rviz` . Now you would be able to see the rviz plugin built by the voxblox package

closed 2020-03-09 by aaravrav142

Hao-Starrr, 2022-11-09:
@nachovizzo Thanks! It is helpful. 
@Tokymin I have the visualization, but it can only display the message after rviz get prepared. If I hide the map, and displace it again, the previous map will not occur. So make sure you set up rviz first and then run the launch file.

Xue1iang, 2023-09-16:
Source /workspace/devel/setup.bash doesn't help......VoxbloxMesh in Rviz sill shows nothing.