ethz-asl/voxblox #308: Build map with sparse feature pintcloud

nhk035, 2019-12-20:
Hi,
I noticed that all sensors you used can generate a dense pointcloud. However, is there any possibility to build a map with sparse point cloud?  For example, I have a camera, which can only generate 2000 3D points(triangulated from orb feature points)  per frame and i know the 6Dof pose of each frame.
Thanks.

mfehr, 2019-12-24:
Hi @nhk035 
We have actually used voxblox in combination with sparse 3D features before. Maybe this helps:
[paper](https://ieeexplore.ieee.org/abstract/document/8460641)
[landmark integration code](https://github.com/ethz-asl/maplab/blob/master/interfaces/voxblox_interface/include/voxblox-interface/integration.h#L13]

nhk035, 2019-12-24:
Thanks for your reply @mfehr . I have read this paper before, but there are some details that were not well understood. So you mean i can find Topomap code in maplab?

mfehr, 2019-12-24:
@nhk035 not the whole code, only the part of topomap that takes a maplab map and integrates the visual 3D landmarks into voxblox, i.e. the function I linked above.
(ping @fabianbl)

nhk035, 2019-12-24:
Thanks, I will try the function.

closed 2019-12-24 by nhk035