ethz-asl/voxblox #318: Demo Error

YJonmo, 2020-04-02:
Hi there,

Thanks for this open source work.

I am trying to run the demo as shown in the https://voxblox.readthedocs.io/en/latest/pages/Running-Voxblox.html
I following error with both Basement and the Lady&Cow datasets:

[ERROR] [1585810479.805742349]: Input pointcloud queue getting too long! Dropping some pointclouds. Either unable to look up transform timestamps or the processing is taking too long.

Also I attached the screenshot below. 
I tried both on Ubuntu 16 and 18 and I got the same error. 

Any help please?

Thanks
Yaqub

[image]

mfehr, 2020-04-02:
Hi @YJonmo , did you build the workspace in release or debug mode?

YJonmo, 2020-04-02:
I followed this link which looks like to be released mode

https://voxblox.readthedocs.io/en/latest/pages/Installation.html

mfehr, 2020-04-02:
You can check if you do `catkin config` in your workspace

YJonmo, 2020-04-02:
Yes it is release:

catkin config --cmake-args -DCMAKE_BUILD_TYPE=Release

YJonmo, 2020-04-02:
Sorry if I sound rude, but has the Voxblox been tested on the Ubuntu recently?

mfehr, 2020-04-02:
Did you do a clean rebuild after switching to release:
```
catkin clean --yes
catkin build voxblox_ros
```

mfehr, 2020-04-02:
We use it often and on many platforms, mostly on Ubuntu 18,04.

alexmillane, 2020-04-02:
Does that error come up more than once? It's very possible that at the start of a dataset the transforms are unavailable as things startup. If the error appears once at the start, and then never again; don't worry.

Regarding the question testing question. I run voxblox on Ubuntu all the time. I'm running it for something else as we speak.

alexmillane, 2020-04-02:
It's quite possible that everything is operating as expected. From your screenshot, the data is being integrated into the map as it should. During your 4h run, it not unlikely that voxblox is just sitting inactive for the 3 hours 58 minutes after the dataset has finished.

YJonmo, 2020-04-02:
Thanks for the replies.

For Ubuntu 18 I am running following for installation:

mkdir -p ~/catkin_ws/src
cd ~/catkin_ws
catkin init
**catkin clean --yes**
catkin config --extend /opt/ros/kinetic
catkin config --cmake-args -DCMAKE_BUILD_TYPE=Release
catkin config --merge-devel

cd ~/catkin_ws/src/
git clone https://github.com/ethz-asl/voxblox.git
wstool init . ./voxblox/voxblox_https.rosinstall
wstool update

cd ~/catkin_ws/src/
catkin build voxblox_ros

When I run init I get error, which I guess it is due to previous installations? 
wstool init . ./voxblox/voxblox_https.rosinstall ==> Error: There already is a workspace config file .rosinstall at ".". Use wstool install/modify.

Sorry I am a Python sort of person and a bit new to ROS and C++ based tools. 
@alexmillane  Yes it only appear once at the beginning but then the code runs forever and never stops, all what I see on screen is below image. I was waiting for 4 hours and it did not finish. How long is it supposed to take to finish the basement dataset demo?

[image]

YJonmo, 2020-04-02:
is it something to do with     <param name="update_mesh_every_n_sec" value="1.0" /> 
I changed the 1.0 to 0.1 and now the Updating mesh is going much faster. But I don't know if it impacts the output.

alexmillane, 2020-04-02:
The "cow and lady" dataset is 2:22s in length. You actually ran the dataset successfully from what I can tell; after the dataset is finished voxblox just sits around waiting for more data.

The output on the terminal is the meshing thread updating the visualization to reflect changes in the voxblox map. There are no changes, so it's just doing nothing.

 So from what I can see, there's no issue here. Was there something else you were expecting?

YJonmo, 2020-04-02:
Interesting, but then where is the output mesh result?
Based on launch file it should be in the mesh_result folder. There is nothing there. 

    <param name="mesh_filename" value="$(find voxblox_ros)/mesh_results/$(anon basement).ply" />

alexmillane, 2020-04-02:
So there are two options. One is to view the mesh in RVIZ, which is probably worth doing. The other is the file to which you refer. To tell voxblox to save a mesh you call the service `generate_mesh`.

alexmillane, 2020-04-02:
I think one source of misunderstanding here is that voxblox (at least in this case) is designed as a _server_. It just sits there and consumes data as it comes in and produces stuff you ask it to. This is in contrast to other programs which run through data, produce outputs and close.

YJonmo, 2020-04-02:
Thanks for calrifying, I will give it a try tomorrow.  

Sorry to ask, by 2:22 s do you mean 2 minutes and 22 s or 2 hours and 22s?

alexmillane, 2020-04-02:
Two _minutes_, sorry. And it is played at real time so the dataset **will** finish in ~2 mins regardless of what is happening on the voxblox end.

I will close this issue for now as I am fairly confident voxblox is operating as expected. Feel free to continue to comment if you need help.

closed 2020-04-02 by alexmillane