ethz-asl/voxblox #320: error while doing ''catkin build"

adbidwai, 2020-04-04:
My operating system is Ubuntu 18.04 and ros distro is melodic. 
While building my workspace I am having the following error:

`fatal error: voxblox/Block.pb.h: No such file or directory`

I have installed all the dependencies and this is the only issue while building the package voxblox_ros
Help is appreciated.
TIA

mzahana, 2020-04-04:
I am facing the same issue but on Jetson TX2. It compiles fine on normal PC.

mfehr, 2020-04-04:
If you very recently updated your master, there has been a change in the way voxblox deals with protobuf messages. Have a look at this PR: https://github.com/ethz-asl/voxblox/pull/313

First, make sure the dependencies (protobuf_catkin) has the right version with `wstool update`.

Second, one of these should solve this, try them in this order:

1. Build with `--force-cmake`:
```
catkin build voxblox_ros --force-cmake
```

2. Clean and rebuild
```
catkin clean --yes
catkin build voxblox_ros
```

3. Set up your workspace again :)

mzahana, 2020-04-04:
It compiled with me after I reconfigured my `catkin_ws`
```
catkin config --merge-devel
```

adbidwai, 2020-04-04:
@mfehr In my workspace I already have packages which I have built with catkin build. Can I build the package by any other method than cleaning my build space and again building using catkin_make?

adbidwai, 2020-04-04:
Actually this is such a package which gets built using catkin_make and not by catkin build, quite weird!

mfehr, 2020-04-04:
@adbidwai that's what I suggested to do, do them in this order, but stop if one works. The first one does not clean.
@mzahana The install instructions of voxblox specify this layout. Some of the catkinizations of dependencies are sometimes not compatible with the linked workspace, unfortunately. However, if it did work before you updated the newest master, then it is likely that the protobuf refactoring is not compatible anymore with linked workspace, which would be an unfortunate side-effect. Was that the case for you?  @helenol wdyt?

mfehr, 2020-04-04:
@mzahana can you try this PR?
https://github.com/ethz-asl/voxblox/pull/321

adbidwai, 2020-04-08:
Ok so I got the solution. I had to clean my build spaces. Just build voxblox first without ``--force-cmake`` and then build other packages.
So clean the build space using ``catkin clean --yes``, then build Voxblox using                                 ```catkin build voxblox_ros``` and then build other packages using ``catkin build``.
Thanks a lot @mfehr and @mzahana for your help.
Also credits to @mihirk284
As I think the issue is solved, I'll close the issue now.

closed 2020-04-08 by adbidwai