ethz-asl/voxblox #328: catkin build error in ubuntu 18.04

VimalRathod, 2020-06-03:
Hi,

I have tried to build the voxblox files many times and I keep getting the same error. Hope you guys could help.

[```
build] Found '13' packages in 0.0 seconds.                                    
[build] Updating package table.                                                
Starting  >>> catkin_simple                                                    
Finished  <<< catkin_simple                       [ 7.2 seconds ]                                                                                                                                          
Starting  >>> eigen_catkin                                                                                                                                                                                 
Starting  >>> gflags_catkin                                                                                                                                                                                
Starting  >>> protobuf_catkin                                                                                                                                                                              
Starting  >>> voxblox_msgs                                                                                                                                                                                 
Finished  <<< eigen_catkin                        [ 3.9 seconds ]                                                                                                                                          
Finished  <<< protobuf_catkin                     [ 4.6 seconds ]                                                                                                                                          
Finished  <<< voxblox_msgs                        [ 26.1 seconds ]                                                                                                                                         
___________________________________________________________________________________________________________________________________________________________________________________________________________
Warnings   << gflags_catkin:make /home/vimal/catkin_ws/logs/gflags_catkin/build.make.000.log                                                                                                               
make[3]: warning: -jN forced in submake: disabling jobserver mode.
make[3]: warning: -jN forced in submake: disabling jobserver mode.
cd /home/vimal/catkin_ws/build/gflags_catkin; catkin build --get-env gflags_catkin | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
...........................................................................................................................................................................................................
Finished  <<< gflags_catkin                       [ 33.4 seconds ]                                                                                                                                         
Starting  >>> glog_catkin                                                                                                                                                                                  
___________________________________________________________________________________________________________________________________________________________________________________________________________
Warnings   << glog_catkin:make /home/vimal/catkin_ws/logs/glog_catkin/build.make.000.log                                                                                                                   
configure.ac:19: installing './compile'
configure.ac:13: installing './missing'
Makefile.am: installing './depcomp'
parallel-tests: installing './test-driver'
make[3]: warning: -jN forced in submake: disabling jobserver mode.
In file included from src/logging_unittest.cc:58:0:
src/googletest.h:579:33: warning: dynamic exception specifications are deprecated in C++11 [-Wdeprecated]
 void* operator new(size_t size) throw(std::bad_alloc) {
                                 ^~~~~
src/googletest.h:586:35: warning: dynamic exception specifications are deprecated in C++11 [-Wdeprecated]
 void* operator new[](size_t size) throw(std::bad_alloc) {
                                   ^~~~~
src/logging_unittest.cc: In function ‘void TestDCHECK()’:
src/logging_unittest.cc:568:3: warning: ‘template<class> class std::auto_ptr’ is deprecated [-Wdeprecated-declarations]
   auto_ptr<int64> sptr(new int64);
   ^~~~~~~~
In file included from /usr/include/c++/7/bits/locale_conv.h:41:0,
                 from /usr/include/c++/7/locale:43,
                 from /usr/include/c++/7/iomanip:43,
                 from src/logging_unittest.cc:44:
/usr/include/c++/7/bits/unique_ptr.h:51:28: note: declared here
   template<typename> class auto_ptr;
                            ^~~~~~~~
In file included from /usr/include/c++/7/ext/hash_set:60:0,
                 from src/glog/stl_logging.h:73,
                 from src/stl_logging_unittest.cc:62:
/usr/include/c++/7/backward/backward_warning.h:32:2: warning: #warning This file includes at least one deprecated or antiquated header which may be removed without further notice at a future date. Please use a non-deprecated interface with equivalent functionality instead. For a listing of replacement headers and interfaces, consult the file backward_warning.h. To disable this warning use -Wno-deprecated. [-Wcpp]
 #warning \
  ^~~~~~~
In file included from src/symbolize_unittest.cc:41:0:
src/googletest.h:579:33: warning: dynamic exception specifications are deprecated in C++11 [-Wdeprecated]
 void* operator new(size_t size) throw(std::bad_alloc) {
                                 ^~~~~
src/googletest.h:586:35: warning: dynamic exception specifications are deprecated in C++11 [-Wdeprecated]
 void* operator new[](size_t size) throw(std::bad_alloc) {
                                   ^~~~~
In file included from src/demangle_unittest.cc:41:0:
src/googletest.h:579:33: warning: dynamic exception specifications are deprecated in C++11 [-Wdeprecated]
 void* operator new(size_t size) throw(std::bad_alloc) {
                                 ^~~~~
src/googletest.h:586:35: warning: dynamic exception specifications are deprecated in C++11 [-Wdeprecated]
 void* operator new[](size_t size) throw(std::bad_alloc) {
                                   ^~~~~
In file included from src/utilities_unittest.cc:32:0:
[... 38 lines cut ...]
Starting  >>> voxblox_rviz_plugin                                                                                                                                                                          
Finished  <<< voxblox_rviz_plugin                 [ 36.5 seconds ]                                                                                                                                         
Starting  >>> voxblox_ros                                                                                                                                                                                  
___________________________________________________________________________________________________________________________________________________________________________________________________________
Errors     << voxblox_ros:make /home/vimal/catkin_ws/logs/voxblox_ros/build.make.000.log                                                                                                                   
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
make[2]: *** [CMakeFiles/voxblox_ros.dir/src/simulation_server.cc.o] Error 4
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/voxblox_ros.dir/all] Error 2
make: *** [all] Error 2
cd /home/vimal/catkin_ws/build/voxblox_ros; catkin build --get-env voxblox_ros | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
...........................................................................................................................................................................................................
[... cut ...]
[build]   Failed:    1 packages failed.                                                                                                                                                                    
[build] Runtime: 7 minutes and 0.3 seconds total.                                                                                                                                                          
[build] Note: Workspace packages have changed, please re-source setup files to use them.
```

dave992, 2020-06-18:
> c++: internal compiler error: Killed (program cc1plus)

That seems to indicate it to be a bug in the compiler itself and not the package it is trying to compile.

kgreenek, 2020-06-18:
I have seen that error before. In my case, it has been caused by running out of ram. Catkin doesn't do anything smart to watch ram usage, but you can reduce the number of concurrent threads it's allowed to use with the `-jN` flag. For example, to limit it to one core pass `-j1` to the call to catkin. Hope that helps!