ethz-asl/voxblox #330: ROS Noetic

UniBwTAS, 2020-06-29:
Will there be a release for ROS Noetic? We tried to simply build with Noetic but got quite a lot error messages.

UniBwTAS, 2020-07-01:
We figured out how to build it successfully. It was sufficient just to build the package `voxblox_ros` with C++14. At least no it can be built. We will test whether it is running correctly.

BananaGorilla98, 2020-09-06:
Hi. May I know how you build the package `voxblox_ros`? I am using ROS Noetic and facing problem while building this package.

alexmillane, 2020-09-07:
Did you try building with c++14 as commented above?

BananaGorilla98, 2020-09-07:
> Did you try building with c++14 as commented above?

I tried to install the C++14 compiler but my computer clashed. Will that out later.

alexmillane, 2020-09-07:
What compiler are you using? If you are using ROS noetic, and therefore presumably on Ubuntu 20.04, gcc (preinstalled) will compile c++14 no issues. In fact, I would have assumed that any non-ancient compiler would compile c++14.

BananaGorilla98, 2020-09-07:
I am using Ubuntu 20.04 default compiler. But it failed to compile. That is why I was trying to install C++14 compiler.

alexmillane, 2020-09-07:
You have a c++14 compiler preinstalled. You need to change the flag 

`-std=c++11`

to 

`std=c++11`

in the line here:

https://github.com/ethz-asl/voxblox/blob/master/voxblox_ros/CMakeLists.txt#L7

@UniBwTAS Is this the only change you had to make? :)

BananaGorilla98, 2020-09-07:
Ok, thanks. :) Will try it out

BananaGorilla98, 2020-09-07:
> You have a c++14 compiler preinstalled. You need to change the flag
> 
> `-std=c++11`
> 
> to
> 
> `std=c++14`
> 
> in the line here:
> 
> https://github.com/ethz-asl/voxblox/blob/master/voxblox_ros/CMakeLists.txt#L7
> 
> @UniBwTAS Is this the only change you had to make? :)

It worked! Thank you!

alexmillane, 2020-09-07:
Great! Glad it worked. I will close this issue now

closed 2020-09-07 by alexmillane