ethz-asl/voxblox #331: No mesh via Rviz

LanWu076, 2020-06-30:
I can see messages received via Rviz but there is no mesh showing on my screen. I'm using cow and lady dataset. The fixed frame is "world". I do have a warning said: No tf data. Actual error:Fixed Frame [world] does not exist. 
Does anybody know what could be wrong?

[image]

tyxxj, 2020-07-30:
Hello, try to source your workspace first on the terminal that you will run rviz

guoxiaoyang12, 2020-10-21:
Have you fixed this problem? I have same porblem

apache8080, 2020-10-21:
I am running into the same issue with every launch file except for basement.

closed 2020-10-25 by LanWu076