ethz-asl/voxblox #341: why the calculation of 'sdf' is different from the paper.

mcfrain, 2020-10-16:
i am confused of the calculation of 'sdf‘, because its different from the paper "Voxblox: Incremental 3D Euclidean Signed Distance Fields for On-Board MAV Planning".
And it seems that the formula on the paper is more easier to understand.
--------------------------------------
--------------------------------------
d(x, p, s) = ||p - x||sign((p - x)(p - s))
Given that x is the center position of the
current voxel, p is the position of a 3D point in the incoming
sensor data, s is the sensor origin, and x, p, s ∈ R3
--------------------------------------
--------------------------------------
Does anyone have any ideas?

helenol, 2020-10-16:
Don't remember exactly but I think the paper version is overly simplistic doesn't cover some rare edge cases if the surface boundary is within that voxel (or isn't as accurate as it could be?), so I guess I must have fixed it after (???). So as the comment says we do this instead:
```c++
// To do this, project the voxel_center onto the ray from origin to point G.
// Then check if the the magnitude of the vector is smaller or greater than
// the original distance..
```
First part is the dist_G_V and second part actually gets you the sign.

https://github.com/ethz-asl/voxblox/blob/master/voxblox/src/integrator/tsdf_integrator.cc#L216-L228
Here's the math:
```C++
  const Point v_voxel_origin = voxel_center - origin;
  const Point v_point_origin = point_G - origin;

  const FloatingPoint dist_G = v_point_origin.norm();
  // projection of a (v_voxel_origin) onto b (v_point_origin)
  const FloatingPoint dist_G_V = v_voxel_origin.dot(v_point_origin) / dist_G;

  const float sdf = static_cast<float>(dist_G - dist_G_V);
```
v_voxel_origin = (x - p)
v_point_origin = (s - p)
dist_G = ||x - p||
dist_G_V = ((x - p) dot (s - p)) / || x- p ||

sdf = ||x - p|| - (x-p) dot (s - p) / || x - p ||

Check out https://stackoverflow.com/questions/7023240/projection-of-point-on-line-in-3d or  https://en.wikipedia.org/wiki/Vector_projection for more details on the projection.

Hope this clarifies it a bit.

mcfrain, 2020-10-17:
Thanks for your quick response, i understand that code now.

closed 2020-10-17 by mcfrain